NETMF/netmf-interpreter #4: Build fails

Injac, 2015-03-11:
I followed exactly the steps in the readme, but the code fails to build. I attached a part from the build_sdk output:

Compiler: VS 12 
Building PreSDK ...
Microsoft (R) Build Engine version 12.0.31101.0
[Microsoft .NET Framework, version 4.0.30319.34014]
Copyright (C) Microsoft Corporation. All rights reserved.

C:\Program Files (x86)\MSBuild\12.0\bin\Microsoft.Common.CurrentVersion.targets(991,5): warning MSB3644: The reference assemblies for framework ".NETMicroFramework,Version=v4.4" were not found. To resolve this, install the SDK or Targeting Pack for this framework version or retarget your application to a version of the framework for which you have the SDK or Targeting Pack installed. Note that assemblies will be resolved from the Global Assembly Cache (GAC) and will be used in place of reference assemblies. Therefore your assembly may not be correctly targeted

The reference assemblies Version 4.4 must be installed in order to compile. The readme does not contain this information's.

Have to check on codeplex, or here on github, if a release is downloadable.

smaillet-ms, 2015-03-12:
Can you post the complete log file? That warning is normal and expected, it's an .NET Framework BUILD warning that indicates it can't find the libraries for the specified framework which of course it can't find because you are in the process of building them. Unfortunately we haven't found a way to disable that particular warning short of cloning and modifying the standard .NET MS build targets files (Not a good idea).

Injac, 2015-03-12:
Sure. Here is the link to the file on my OneDrive (a max. of ~26K characters allowed here):

https://onedrive.live.com/redir?resid=e11170f8ecc6005c!27816&authkey=!AJkbSi7NMYUDImw&ithint=file%2ctxt

Injac, 2015-03-12:
I have written a blog-post a while ago, how to compile 4.3 and what the requirements are: 

http://www.awsomedevsigner.com/2015/01/04/successful-compilation-of-the-netmf-4-3-rtm-codeplex-version/

Maybe this helps. Looks like its missing also the WiX assemblies and some dependencies mentioned in my blog-post.

smaillet-ms, 2015-03-12:
Did you install the Bin and TOOLS from the pre-requisites ZIP file? Based on your log file those should be expanded to D:\gitsource\bin and D:\gitsource\tools

That's where the WIX libraries are "installed" We are planning to change the build to use a normal installation point for all of the required binaries, as well as include pre-build testing to report helpful error messages regarding any thing that isn't found.

Injac, 2015-03-12:
Yes, I did. But not in within the directories you mentioned. Let me try that.

smaillet-ms, 2015-03-12:
ok, assuming that works if you have a suggestion on how to explain that better I can update the readme. (I did it with a bit more detail in the wiki, perhaps I should update the readme to just point to the WIKI)

Injac, 2015-03-12:
I think it works now (compiling). Thanks ;) I think a good explanation could be something like:

1. Create a directory like netmf-build-root (or something like that and refer to it as "netmf build root folder")
2. Checkout the source within that directory
3. Download the zip with the tools and copy both folders into the "netmf build root folder" 

And then point the user to the Wiki for details.

Injac, 2015-03-12:
Ok, here is the build-log after putting the tools into the right directory. Still does not complete, but looks much better:

https://onedrive.live.com/redir?resid=e11170f8ecc6005c!27817&authkey=!AHPaxWoragbMIQs&ithint=file%2ctxt

smaillet-ms, 2015-03-12:
What the.... :confused: You wouldn't perchance have setup the Roslyn compiler as your compiler for VS2013 would you?

Injac, 2015-03-12:
I don't think so. I had the Roslyn Preview installed long a go, the Roslyn SDK Project Templates and the Roslyn Syntax vizualizer. All disabled in VS a long time ago.
[image]

smaillet-ms, 2015-03-12:
Might be a bug in their setup where it's still using the compiler. Can you uninstall it? The number of compiler warnings etc.. on code that builds on multiple installations for us and has nothing to do with any dependencies hints at something funny going on with the compiler...

Injac, 2015-03-12:
Yes. Will do that now. Just a second.

Injac, 2015-03-12:
Even after removing it, it uses:

Microsoft\VisualStudio\12.0\Extensions\ce0qbks5.bng\rcsc.exe /noconfig 

within Visual Studio. 

Will check that and let you know about the solution.

Injac, 2015-03-12:
Ok. After the extensions (Roslyn) are uninstalled and the machine rebooted, the tools copied into the right directories, The build will complete. Here is the build log:

https://onedrive.live.com/redir?resid=e11170f8ecc6005c!27818&authkey=!AH1aRfv4-2AXKUc&ithint=file%2ctxt

I will check this log further, to see if there are other issues with my system.

smaillet-ms, 2015-03-12:
looks good. You should see the MSI and VSIX packages in the build output folder. (Which reminds me I should probably add that little detail to the getting started guide :smiley: )

Injac, 2015-03-12:
Yepp, here you go
[image]
 :) And a good idea to describe the output folder a bit.

cw2, 2015-03-12:
To suppress MSB3644: The reference assemblies for framework ".NETMicroFramework,Version=vX.Y" were not found warnings, create FrameworkList.xml in "C:\Program Files (x86)\Reference Assemblies\Microsoft\Framework.NETMicroFramework\vX.Y\RedistList" with the following content:
```
<FileList Name=".NET Micro Framework X.Y.0.0" Redist="Microsoft-Windows-CLRCoreComp-MicroFramework-X.Y.0.0" ToolsVersion="4.0"/>
```
Where X.Y is the desired version.

Injac, 2015-03-12:
Hi! @cw2 Thanks for the hint :) Will do that as soon as I am at home. Currently at work.

Savvkin, 2015-03-13:
Did you see my guide about building the vNext from sources [here](https://www.ghielectronics.com/community/forum/topic?id=18352)? It worked for Codeplex sources, but I didn't test it with current github sources.

Injac, 2015-03-13:
@Savvkin Thanks for the link :) +1 and sharing it on Twitter :) Never to late to learn something.

closed 2015-03-13 by smaillet-ms