NETMF/netmf-interpreter #5: Eliminate or suppress warnings for build_sdk

smaillet-ms, 2015-03-12:
The build process started from build_sdk.cmd will generate a number of warnings:  
#### msbuild warnings
* `warning MSB3644: The reference assemblies for framework ".NETMicroFramework,Version=vX.Y"`
* `warning MSB4011: "<path>.Targets" cannot be imported again. It was already imported at "<path> <line>,<col>)". This is most likely a build authoring error. This subsequent import will be ignored.` 

#### Linker warnings
* `warning LNK4221: This object file does not define any previously undefined public symbols, so it will not be used by any link operation that consumes this library`
*  `warning LNK4199: /DELAYLOAD:<dllname>.dll ignored; no imports found from <dllname>.dll` 
*  `warning LNK4042: object specified more than once; extras ignored` 

#### C++ Compiler warnings
* `warning C4065: switch statement contains 'default' but no 'case' labels`
* `warning C4996: '<apiname>: was declared deprecated`
* ` warning C4018: '<operator>' : signed/unsigned mismatch
* `warning C4244: '<operatpr>' : conversion from '<srctype>' to '<dsttype>', possible loss of data

smaillet-ms, 2015-03-24:
All but one of the MSB4011 were eliminated. One remains due to how the featureproj and libcatproj work along with SolutionWizard. Fixing that would essentially mean a significant modification to SolutionWizard or creating a clone of the OpenSSL featureporj. Neither of which seems like they are worth the effort for an annoying but benign warning.

cw2, 2015-03-24:
If you want, I could take care of the compiler warnings (fix them in NETMF code, suppress in 3-rd party libraries)? So you can work on more important stuff...

smaillet-ms, 2015-03-24:
Works for me! :smile: Key point is "suppress in third party" unless it's actually a bug. Though the third party here is mostly OpenSSL where subtle issues and seemingly innocent "corrections" can cause large security holes so suppressing rather than altering the code is generally preferred.

smaillet-ms, 2015-03-26:
Morteza got the remaining MSB4011 with #39

closed 2015-03-30 by smaillet-ms