NETMF/netmf-interpreter #10: STM32F4 USART is broken

Savvkin, 2015-03-13:
In the file `STM32F4_usart_functions.cpp` at [Line:112](https://github.com/NETMF/netmf-interpreter/blob/dev/DeviceCode/Targets/Native/STM32F4/DeviceCode/STM32F4_USART/STM32F4_usart_functions.cpp#L112) you decrement the `ComPortNum` value and then call `CPU_USART_GetPins` at [Line:182](https://github.com/NETMF/netmf-interpreter/blob/dev/DeviceCode/Targets/Native/STM32F4/DeviceCode/STM32F4_USART/STM32F4_usart_functions.cpp#L182).
Inside `CPU_USART_GetPins` at [Line:465](https://github.com/NETMF/netmf-interpreter/blob/dev/DeviceCode/Targets/Native/STM32F4/DeviceCode/STM32F4_USART/STM32F4_usart_functions.cpp#L465) you decrement the `ComPortNum` again.

smaillet-ms, 2015-03-13:
Yep, that whole virtual port for the ITM has turned out to be one of those "seemed like a good idea at the time" things... Probably the best option is just rolling that back out. The whole Communications redirector and multiple xxx_printf variants in the HAL/PAL is in need of some heavy refactoring/re-design as plugging in the ITM channel should be a brain dead simple thing.

smaillet-ms, 2015-07-07:
Fixed in #223 and #222

closed 2015-07-07 by smaillet-ms