NETMF/netmf-interpreter #11: STM32F411 is not supported

Savvkin, 2015-03-13:
`platform_selector.h` doesn't allow to configure STM32F411 to run at stock 100 MHz frequency (Unable to set `100000000` value to `SYSTEM_CLOCK_HZ` and `SYSTEM_CYCLE_CLOCK_HZ`), so it's not possible to run STM32F411 at full speed.

Here is the original C++ configuration from the STM32F411 SDK sample:
```
SYSCLK(Hz)        = 100000000
HCLK(Hz)          = 100000000
AHB Prescaler     = 1
APB1 Prescaler    = 2
APB2 Prescaler    = 1
HSI Frequency(Hz) = 16000000
PLL_M             = 16
PLL_N             = 400
PLL_P             = 4
PLL_Q             = 7
```

smaillet-ms, 2015-03-13:
Can you be more specific on what the issue is here? Platform_selector.h is (as the name implies) platform specific, so you can set whatever values you need or want.

Savvkin, 2015-03-13:
If you try to set `SYSTEM_CLOCK_HZ` to `100000000` you get compilation error in `STM32F4_bootstrap.cpp` at [Line:68](https://github.com/NETMF/netmf-interpreter/blob/dev/DeviceCode/Targets/Native/STM32F4/DeviceCode/STM32F4_Bootstrap/STM32F4_bootstrap.cpp#L68), because the code that calculates PLL doesn't expect 100MHz frequency.

smaillet-ms, 2015-03-13:
Excellent, thanks for the quick response. That's pretty clear and actionable. If you've got a fix, please fill out a [Contribution License Agreement (CLA)](Contribution License Agreement) if you haven't already and post a pull request.

josesimoes, 2015-04-21:
Setting sys clock @ 100Mhz won't work if you need to use the UART and USB. With that frequency isn't possible to get proper PPL values in order to achieve the standard baud rates and standard USB clock and bus timings. For everything else I would say that it should work at any sys clock.

Savvkin, 2015-04-22:
I've tested the settings above by creating uVision app and deploying it to Nucleo-F411RE board and can confirm that UART works. I can't say anything about USB, because board doesn't have it so I don't need it.
I thinks it's a bug, there should be more flexisble way to set Clock and PLL values, than it currently exists in netmf. For example, override automatic configuration and allow manually set Clock/PLL values.
Not every application requires USB. Why should we trade speed for USB that not even wired? Even if USB exists, we can boot underclocked with USB enabled only when we need it.

piwi1263, 2015-04-22:
Hi savvkin,

I just got netmf 4.3 running on a Nucleo-F411RE by not using the regular
USB but via the Morpho connectors, using PA11, PA12 connected to a USB
cables D+, D- and used the VBUS as well all without using the ST-Link USB
Connector. Works OK for me. One can even cutoff the ST-Link part of the PCB
but need to solder a crystal X3 marked on the PCB.

Look here for more details ->
https://www.ghielectronics.com/community/forum/topic?id=18762&page=2#msg188043

Hope this helps.

Kind regards,

Peter

Savvkin, 2015-04-22:
Peter, the bug report is about running STM32F411 at stock clock frequency, which is 100 MHz.
Did you run it at 100 MHz?

piwi1263, 2015-04-22:
I never changed anything about the clock, so I recon it is running at the frequency the product is default setup for. At the moment it one runs with the ST-Link PCB attached and taking the clock from that art of hte PCB.

Savvkin, 2015-04-22:
What is in your platform_selector.h file? I bet it's
```Cpp
#define SYSTEM_CLOCK_HZ                 72000000  // 72MHz
```

josesimoes, 2015-04-22:
@Savvkin: I can't find a valid clock configuration to run SYS clock @ 100MHz... The closest I can get is 96MHz. 
Like Peter I'm running 4.3.1 in a Nucleo 411 board. I do have an 8MHz crystal fitted and I have USB connection and UART.

Please understand that I'm just commenting about the SYS clock values. No complaints about your suggestion to have more flexible clock configurations.

piwi1263, 2015-04-22:
@savvkin: Forget about my comments, I was only reacting on your entry stating no USB was avail on the board. 

I've only got pre-build binaries (for 4.3 and NOT 4.4) without the sources, but the frequency has been set to 96 MHz with an 8 MHz crystal. For me that's close enough to 100 MHz.

smaillet-ms, 2015-05-13:
Am I correct in understanding this is a resolved issue? (e.g. 96Mhz works in hardware and close enough to the desired frequency that no additional work is required)

closed 2015-05-21 by smaillet-ms

luca-saggese, 2015-07-27:
Hello josesimoes,
could you share your F411 Solution folder?
i was able to compile it with 96MHz clock, but i'm missing something as the complete thinibooter does not work.

thanks
Luca

josesimoes, 2015-07-27:
Hi,

Sure. I'm out right now and will be back to the office next week. I'll send you that then.

Cheers,

luca-saggese, 2015-07-27:
thanks you!
will wait anxiously :)

josesimoes, 2015-08-03:
Hi Luca,

Can't send an attachment trough GitHub messaging... Please email me your address at jose DOT simoes AT eclo DOT pt.