NETMF/netmf-interpreter #18: Files that do not exist in the TinyCLR_server.sln have been defined.

Sirokujira, 2015-03-16:
It has been described as follows in TinyCLR_server.sln.

<a href="https://github.com/NETMF/netmf-interpreter/blob/dev/TinyCLR_server.sln#L20">Line:20</a>
<a href="https://github.com/NETMF/netmf-interpreter/blob/dev/TinyCLR_server.sln#L25">Line:25</a>

The following file does not exist.
Framework\Tools\ConnectionManager\ConnectionManager.csproj
Framework\Tools\ConnectionManagerHost\ConnectionManagerHost.csproj

smaillet-ms, 2015-03-16:
TinyCLR_Server.sln isn't really a supported solution, you can't build it so at best it was a code navigation tool, but over the years was not maintained or kept up. At this point it's debatable whether just deleting it or spending the effort to figure out all the missing content (or extra content) etc.. as it will always be a manual process until such time as we have shifted the build to where we can build the framework directly in VS.

Sirokujira, 2015-03-17:
Thank you for your answer.

closed 2015-03-17 by Sirokujira