NETMF/netmf-interpreter #29: Duplicate device startup files

cw2, 2015-03-20:
There are multiple copies of device startup assembly files (bootvector.s, firstentry*, vectorshandlers.s, vectorstrampolines.s etc.) present in the source tree, some of which are no longer relevant for Cortex-M. Would it be possible to consolidate those files or perhaps replace them with CMSIS variants?

smaillet-ms, 2015-03-20:
Yes, that's part of the plan for the CMSIS support I mentioned. We've pruned a large portion of the legacy code but there's still more to go.

smaillet-ms, 2015-06-17:
This has been cleaned up a bit in the current tree, though there are still duplicates for the OS based builds that only differe by a small factor. These should eventually be cleaned up such that only one copy is needed.

lt72, 2015-07-14:
Moving to 4.5, as this issue will be impacted by the next version of mBed, which we have no visibility into yet.