NETMF/netmf-interpreter #30: missing vsix

fredrik-ngenic, 2015-03-20:
Hi,
don't know if this is the right place but...

When i run the build_sdk.cmd I get the sdk msi in the place that is mentioned in the wiki,
but the vsix files aren't where they're supposed to (BuildOutput\public\Release\Server\dll).
And I can't find them anywhere else either...
Any ideas?

Regards 
Fred

smaillet-ms, 2015-03-20:
This is the perfect place to ask such questions!
Sounds like you don't have the Visual Studio SDK installed. The build system will skip generating the VSIX files if it detects the SDK is not installed. Now that we've cleaned up the SDK build it's probably worth generating an informative message on that. (e.g. if the build is skipping a particular VSIX it should print a message to ley you know it is doing so and why)

mortezag, 2015-03-21:
Maybe related to #33

fredrik-ngenic, 2015-03-23:
Thank you,
I'll look into that

closed 2015-03-24 by smaillet-ms