NETMF/netmf-interpreter #33: VSIX generation fails with VSSDK1039: Source.extension.vsixmanifest file not found

maiorfi, 2015-03-21:
I followed building guide but even though building SDK runs successfully, VSIX generation fails with mentioned VSSDK1039 error. How could I verify "correctness" of my VSSDK setup?

Thanks!

maiorfi, 2015-03-21:
Opening Solution located at:

C:\NETMF\netmf-interpreter\Framework\CorDebug\vs12

I can build .vsix, despite I get a couple of:

Error	2	The "Microsoft.SPOT.Tasks.Internal.CreateAssemblyFragment" task could not be loaded from the assembly C:\NETMF\netmf-interpreter\BuildOutput\Public\Debug\Server\dll\Microsoft.SPOT.Tasks.Internal.dll. Could not load file or assembly 'file:///C:\NETMF\netmf-interpreter\BuildOutput\Public\Debug\Server\dll\Microsoft.SPOT.Tasks.Internal.dll' or one of its dependencies. The system cannot find the file specified. Confirm that the <UsingTask> declaration is correct, that the assembly and all its dependencies are available, and that the task contains a public class that implements Microsoft.Build.Framework.ITask.

for Debugger and WinUSB projects.

mortezag, 2015-03-21:
There is most likely a problem in the source-code, due to a bad check-in on our side. We will investigate it asap.

smaillet-ms, 2015-03-21:
You need to at least have built the tools binaries before you can open the solution files for the VS extensions and MFDeply. This is done as part of building the SDK with build.sdk (particularly when it runs msbuild on sdk.dirproj).

smaillet-ms, 2015-03-21:
I just ran a test on my home machine with the latest code checked in and the VSIX packages were built OK. Do you have VS2013 and the VS2013 SDK installed?

cw2, 2015-03-21:
I can reproduce the issue on my workstation, VS2013 and VS2013 SDK installed. The information in the log indicates the source.extension.wixmanifest is searched for in cordebugvs12.csproj (as per the original post), while it is present in NetmfVS2013.csproj.

I guess the cause is the discrepancy between conditional includes in `Support\frameworkbase\dotNetMF.proj`:
`<Project Condition="Exists('$(VSSDK12INSTALLDIR)')" Include="$(SPOCLIENT)\Framework\CorDebug\vs12\CorDebugvs12.csproj"/>`
and `setup\ProductSDK\VsixPackages.dirproj`
`<Project Condition="Exists('$(VSSDK12INSTALLDIR)')" Include="$(SPOCLIENT)\Framework\CorDebug\vs12\netmfvs2013.csproj" />`

smaillet-ms, 2015-03-22:
There's no discrepancy in those conditionals. One is conditional testing for the SDK before building the actual VS package binaries the other is testing for VSSDK before generating the actual  VSIX package, which is what they should be doing. I've got two machines where this all works so something else is going on. Before moving into GitHub we had race conditions in the build where some dependencies weren't made explicit. Perhaps we have another instance of that here... Can you post the full log file?

maiorfi, 2015-03-22:
I just uploaded my latest build_sdk logs at following address (there are several building sessions inside):

https://dl.dropboxusercontent.com/u/7414592/Embedded/NETMF44_Binaries/build_sdk_logs.7z

Thanks.

cw2, 2015-03-22:
I have compared logs from successful and failed build - the difference is that in the successful case `CorDebugvs12.csproj` has missing `CopyVsixManifestToOutput` property, as result of `CreateVsixContainer = False` and `DeployExtension = False` (the value of `CopyVsixManifestToOutput` is calculated in
`C:\Program Files (x86)\MSBuild\Microsoft\VisualStudio\v12.0\VSSDK\Microsoft.VsSDK.targets`
as
`<CopyVsixManifestToOutput Condition="'$(CopyVsixManifestToOutput)' == '' and ('$(CreateVsixContainer)' != 'false' or '$(DeployExtension)' != 'false' or '$(CopyVsixExtensionFiles)' != 'false')">true</CopyVsixManifestToOutput>`).

The relevant build steps are
```
Done building target "CopyPkgDef" in project "CorDebugvs12.csproj".: (TargetId:1969)
Target "CreateVsixContainer" skipped, due to false condition; ($(CreateVsixContainer)) was evaluated as (False).
Target "DeployVsixExtensionFiles" skipped, due to false condition; ('$(DeployExtension)'=='true') was evaluated as ('False'=='true').
Target "CopyVsixManifestFile" skipped, due to false condition; ('$(CopyVsixManifestToOutput)'=='true') was evaluated as (''=='true').
Target "CopyVsixExtensionFiles" skipped, due to false condition; ('$(CopyVsixExtensionFiles)'=='true' AND '$(CopyVsixExtensionLocation)' != '') was evaluated as ('false'=='true' AND '' != '').
Target "CreateMenuPkgDef: (TargetId:1970)" in file "C:\Program Files (x86)\MSBuild\Microsoft\VisualStudio\v12.0\VSSDK\Microsoft.VsSDK.Common.targets" from project "R:\netmf-interpreter\Framework\CorDebug\vs12\CorDebugvs12.csproj" (target "PrepareForRun" depends on it):
Skipping target "CreateMenuPkgDef" because it has no outputs.
Though the target has declared its outputs, the output specification only references empty properties and/or empty item lists.
Done building target "CreateMenuPkgDef" in project "CorDebugvs12.csproj".: (TargetId:1970)
Target "PrepareForRun: (TargetId:1971)" in file "C:\Program Files (x86)\MSBuild\12.0\bin\Microsoft.Common.CurrentVersion.targets" from project "R:\netmf-interpreter\Framework\CorDebug\vs12\CorDebugvs12.csproj" (target "CoreBuild" depends on it):
Done building target "PrepareForRun" in project "CorDebugvs12.csproj".: (TargetId:1971)
Target "UnmanagedRegistration" skipped, due to false condition; ('$(RegisterForComInterop)'=='true' and '$(OutputType)'=='library') was evaluated as ('false'=='true' and 'Library'=='library').
Target "IncrementalClean: (TargetId:1972)" in file "R:\netmf-interpreter\tools\Targets\Microsoft.SPOT.Build.Common.Targets" from project "R:\netmf-interpreter\Framework\CorDebug\vs12\CorDebugvs12.csproj" (target "CoreBuild" depends on it):
Done building target "IncrementalClean" in project "CorDebugvs12.csproj".: (TargetId:1972)
Target "_VsixCleanRecordFileWrites: (TargetId:1973)" in file "C:\Program Files (x86)\MSBuild\Microsoft\VisualStudio\v12.0\VSSDK\Microsoft.VsSDK.Common.targets" from project "R:\netmf-interpreter\Framework\CorDebug\vs12\CorDebugvs12.csproj" (target "CoreBuild" depends on it):
Task "ReadLinesFromFile" (TaskId:1383)
  Task Parameter:File=R:\netmf-interpreter\BuildOutput\public\Release\Server\obj\Microsoft.SPOT.Debugger.CorDebug.12\__COMMON\CorDebugvs12.csproj.VsixDeployedFileListAbsolute.txt (TaskId:1383)
Done executing task "ReadLinesFromFile". (TaskId:1383)
Task "RemoveDuplicates" (TaskId:1384)
Done executing task "RemoveDuplicates". (TaskId:1384)
Task "MakeDir" (TaskId:1385)
  Task Parameter:Directories=R:\netmf-interpreter\BuildOutput\public\Release\Server\obj\Microsoft.SPOT.Debugger.CorDebug.12\__COMMON\ (TaskId:1385)
Done executing task "MakeDir". (TaskId:1385)
Task "WriteLinesToFile" (TaskId:1386)
  Task Parameter:File=R:\netmf-interpreter\BuildOutput\public\Release\Server\obj\Microsoft.SPOT.Debugger.CorDebug.12\__COMMON\CorDebugvs12.csproj.VsixDeployedFileListAbsolute.txt (TaskId:1386)
  Task Parameter:Overwrite=True (TaskId:1386)
Done executing task "WriteLinesToFile". (TaskId:1386)
Done building target "_VsixCleanRecordFileWrites" in project "CorDebugvs12.csproj".: (TargetId:1973)
Target "PostBuildEvent" skipped, due to false condition; ('$(PostBuildEvent)' != '' and ('$(RunPostBuildEvent)' != 'OnOutputUpdated' or '$(_AssemblyTimestampBeforeCompile)' != '$(_AssemblyTimestampAfterCompile)')) was evaluated as ('' != '' and ('OnBuildSuccess' != 'OnOutputUpdated' or '' != '')).
Target "CoreBuild: (TargetId:1974)" in file "C:\Program Files (x86)\MSBuild\12.0\bin\Microsoft.Common.CurrentVersion.targets" from project "R:\netmf-interpreter\Framework\CorDebug\vs12\CorDebugvs12.csproj" (target "Build" depends on it):
Done building target "CoreBuild" in project "CorDebugvs12.csproj".: (TargetId:1974)
Target "CreateTfsBuildInfoResource" skipped, due to false condition; ( $(AddBuildInfoToAssembly)==true ) was evaluated as ( false==true ).
Target "AfterBuild: (TargetId:1975)" in file "C:\Program Files (x86)\MSBuild\12.0\bin\Microsoft.Common.CurrentVersion.targets" from project "R:\netmf-interpreter\Framework\CorDebug\vs12\CorDebugvs12.csproj" (target "Build" depends on it):
Done building target "AfterBuild" in project "CorDebugvs12.csproj".: (TargetId:1975)
Target "_CheckForCompileOutputs" skipped. Previously built successfully.
```

In the failed case `CopyVsixManifestToOutput` is true, because both `CreateVsixContainer` and `DeployExtension` are also true and the build steps are:
```
Done building target "CopyPkgDef" in project "CorDebugvs12.csproj".: (TargetId:1969)
[... cut ...]
C:\Program Files (x86)\MSBuild\Microsoft\VisualStudio\v12.0\VSSDK\Microsoft.VsSDK.targets(59,5): error VSSDK1039: Source.extension.vsixmanifest file not found in project. If a file with this name is present in the project, make sure the build action is set to "None". [R:\netmf-interpreter\Framework\CorDebug\vs12\CorDebugvs12.csproj]
Done executing task "FindVsixManifest" -- FAILED. (TaskId:1383)
Done building target "FindSourceVsixManifest" in project "CorDebugvs12.csproj" -- FAILED.: (TargetId:1970)
Target "_CheckForCompileOutputs" skipped. Previously built successfully.
```

Unfortunately, I have not yet found the reason why those two properties have different values.

cw2, 2015-03-22:
I forgot to mention the build breaks after commit ab03f51e7f3eb17a255720c4c9c3e479a5a7a219.

cw2, 2015-03-22:
It seems there is a circular dependency between `FLAVOR_WIN` and `Configuration`:

BuildEnv.props
`<FLAVOR_WIN Condition="'$(FLAVOR_WIN)'==''">$(Configuration)</FLAVOR_WIN>`

cordebugvs12.csproj
`<Configuration Condition="'$(FLAVOR_WIN)'!=''">$(FLAVOR_WIN)</Configuration>`

During the failed build `Configuration = Release`, but `FLAVOR_WIN` is empty.

closed 2015-03-24 by smaillet-ms