NETMF/netmf-interpreter #45: CorDebug build fails when VSSDK is in ..\tools

cw2, 2015-03-26:
When v11 (2012) or v14 (2015) VSSDK is not installed on the machine, but present in `..\tools\x86\MicrosoftSDKs\VSSDK\VS##` the build of `...\vs##\CorDebug.csproj`  fails due to missing `C:\Program Files (x86)\MSBuild\Microsoft\VisualStudio\v##.0\VSSDK\Microsoft.VsSDK.targets`.

This is caused by change in `Microsoft.SPOT.CSharp.Host.Targets` in commit ad67ceb389eadd6535bb30feb637d77f3d81bc96, where initialization of `VSToolsPath` variable was added. Before that, MSBuild files for VSSDK 2013 were always used.

The proposed fix:
1. Include VSSDK MSBuild files in the respective `..\tools\...\VSSDK\VS##\MSBuild` directories (not in the zip download, has to be copied manually)
2. In `setenv_base.cmd` add `if EXIST "%VSSDK##INSTALLDIR%" set VSToolsPath="%VSSDK##INSTALLDIR%\MSBuild"`
3. In `Microsoft.SPOT.CSharp.Host.Targets` add conditions `<VSToolsPath Condition="'$(VSToolsPath)'==''"`

cw2, 2015-03-26:
Then it would be possible to replace
```
<VisualStudioVersion Condition="'$(VisualStudioVersion)' == ''">11.0</VisualStudioVersion>
<VSToolsPath Condition="'$(VSToolsPath)' == ''">$(MSBuildExtensionsPath32)\Microsoft\VisualStudio\v$(VisualStudioVersion)</VSToolsPath>
```
with
```
<NetMFBuildVsSdkVersion>11</NetMFBuildVsSdkVersion>
```
and delete `<VSToolsPath>` in .csproj files where the specific version is not set (?)

cw2, 2015-03-26:
`VSToolsPath` has to be version-specific in setenv_base.cmd to avoid overwriting:

2. In `setenv_base.cmd` add `if EXIST "%VSSDK##INSTALLDIR%" set VSSDK##ToolsPath=%VSSDK##INSTALLDIR%MSBuild`
3. In `sadf` add conditions `<VSToolsPath Condition="'$(VSSDK##ToolsPath)' == ''">$(MSBuildExtensionsPath32)\Microsoft\VisualStudio\v##.0</VSToolsPath>`
where ## is 11, 12, 14.

cw2, 2015-03-26:
Well, it seems the issue is more complex than I thought, the VSSDK tools path is used at more places in the build, so the above fix does not cover it all. I have to think it through a little bit more or hope someone else comes with a better solution. Sorry for the comment and edit flood :blush:

Also, I am starting to think whether this is an issue worth fixing - perhaps it would be much easier just to expect VSSDK files in their standard install locations and remove checks for `..\tools` tree, or just make directory links there...

cw2, 2015-03-26:
As it turned out, the fix for path is relatively easy, but the build fails later on with `error MSB6001: Invalid command line switch for "CreatePkgDef.exe". Can not find the tools for VS SDK.`. CreatePkgDef   task checks several environment variables and registry keys and it seems there is no easy way to make it work correctly when VSSDK is not installed and only present in `..\tools` directory, unless there is either `VsSDKToolsPath` or `VsSDKInstall` environment variable set or `InstallDir` value in VisualStudio or VSIP keys.

If I understand it correctly, executing VSSDK tasks from within `..\tools` directory now works only for v12 (2013); assembly references work fine for all VSSDK versions. The fix would require setting environment variables in MSBuild (temporarily, because they are VSSDK version-specific ?) - it is much easier to simply set the respective `InstallDir` values for VSSDKs on a computer where they are not installed in a standard way (just copied into Program Files or `..\tools`).

Included for completeness is the list of checks performed by CreatePkgDef task:
1. Registry value `HKLM\Software\Microsoft\VisualStudio\##\InstallDir`
If not empty use path + "createpkgdef.exe" if it exists, otherwise continue
2. Environment variable `VsSDKToolsPath`
3. Environment variable `VsSDKInstall`
4. Registry value `HKLM\SoftwareMicrosoft\VisualStudio\VSIP\##\InstallDir`
5. Path from 4. + `"\VisualStudioIntegration\Tools\bin"`
Use first non-empty value from 2. - 5. + "createpkgdef.exe".

smaillet-ms, 2015-03-26:
building from the \tools path is no longer supported, ultimately we want to eliminate the use of the ..\bin and ..\tools completely

closed 2015-03-26 by cw2