NETMF/netmf-interpreter #56: StringTable update and cleanup

cw2, 2015-03-31:
The string table (`c_CLR_StringTable_Data[]`, `c_CLR_StringTable_Lookup[]`) in `CLR\Core\StringTable.cpp` was generated from assemblies that do not exist anymore and its content is outdated (~20%, cca 3 KB). Also, the mechanism used to create it has changed and there is some obsolete code and unused files.

So, I'd suggest to
1. Modify `MetaDataProcessor.exe` to exclude strings such as `$$method0x6000*` and `<PrivateImplementationDetails>{GUID}` from generated string table,
2. Update list of `StringTableInputs` in `Microsoft.SPOT.Support.Settings` to include valid *.strings,
3. Generate new string table and update `StringTable.cpp` content,
4. Delete obsolete `tools\scripts\compile_stringtable.cmd`,
5. Delete code in `#if defined TINYCLR_CREATE_STRINGTABLE`.

cw2, 2015-03-31:
Perhaps a better alternative to 1. (+ 3.) is to create an inline msbuild task directly in `Microsoft.SPOT.Support.Settings` (?)

cw2, 2015-07-15:
I volunteer for this issue, I guess there is a lot of more important things you have to do...

lt72, 2015-07-21:
Metadataprocessor generates the string table and it happens through the -loadStrings and -generateStringsTable switches.  The obsolete script  should be updated with a fresh list of assemblies but having a dedicated task would actually be ideal.
Whatever we do, the important bit is to document and log clearly, before or after the build process, in a script or in a task, which assemblies are we including in the StirngTable generation.
Last, the more assemblies one includes, the more strings you get. More strings, faster perf but also larger code size.

example of config script the obsolete script depends on below: 

Cmd>more tools\Libraries\opt_stringtable.cfg
 -loadStrings %BUILD_TREE_CLIENT%\pe\mscorlib.strings
 -loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.DirectBand.Protocol.strings
 -loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.SPOT.Application.Channels.strings
 -loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.SPOT.Application.Data.strings
#-loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.SPOT.Application.Samples.strings
#-loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.SPOT.Application.Tests.strings
 -loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.SPOT.Application.TimeChannel.strings
#-loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.SPOT.Application.WatchFaces.strings
 -loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.SPOT.Data.strings
 -loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.SPOT.Native.strings
#-loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.SPOT.OEM.Citizen.strings
#-loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.SPOT.OEM.Fossil.strings
#-loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.SPOT.OEM.Seiko.strings
#-loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.SPOT.OEM.strings
#-loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.SPOT.OEM.Suunto.strings
 -loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.SPOT.Shell.strings
#-loadStrings %BUILD_TREE_CLIENT%\pe\Microsoft.SPOT.Verification.strings

-generateStringsTable %BUILD_TREE_CLIENT%\Stubs\StringTable.cpp

any contribution is appreciated

Cheers
Lorenzo

cw2, 2015-07-21:
That corresponds with `buildstring` target in `Microsoft.SPOT.Support.Settings`, which generates string table from the following entries:
```
  <ItemGroup>
    <StringTableInputs Include="Microsoft.SPOT.Application.Channels.strings">...</StringTableInputs>
    <StringTableInputs Include="Microsoft.SPOT.Application.Data.strings">...</StringTableInputs>
    <StringTableInputs Include="Microsoft.SPOT.Application.TimeChannel.strings">...</StringTableInputs>
    <StringTableInputs Include="Microsoft.SPOT.Data.strings">...</StringTableInputs>
    <StringTableInputs Include="Microsoft.SPOT.Native.strings">...</StringTableInputs>
    <StringTableInputs Include="Microsoft.SPOT.Shell.strings">...</StringTableInputs>
    <StringTableInputs Include="mscorlib.strings">...</StringTableInputs>
  </ItemGroup>
```