NETMF/netmf-interpreter #64: Upgrade Visual C++ projects to v12 (2013)

cw2, 2015-04-01:
It would be beneficial to have Visual C++ projects upgraded so they can be opened in Visual Studio 2013 (and later) and in several cases they can be used directly in msbuild (e.g. SDK server tools).

If I understand it correctly, it is not needed to convert them manually, it should be possible to modify `MakeVCProject` task in `Framework\Tools\BuildTasksNativeBuild` (or create `MakeVcxProject` task). Alternatively, an XSLT transformation could be used to create .vcxproj from dotnetmf.proj.

smaillet-ms, 2015-04-01:
See also #18. The old VCPROJ files were removed to avoid confusion and the remaining unsupported vcxproj and solutions will be removed soon. At best they were a partial code navigation solution but failed to give a real view of things in the face of the heavily preprocessor configured build used for NETMF. While I'm open to anyone who has the time to offer and maintain a better answer to the problem - supporting something that in my view is half baked at best isn't a high priority for the team.

cw2, 2015-04-02:
Ok, I understand and agree. We'll see what can be done after the changes are finalized.

closed 2015-04-02 by smaillet-ms