NETMF/netmf-interpreter #65: Bit shift operators in VB

dimso, 2015-04-01:
Please add back support for bit shift operators in VB.

This bug is also reported on Codeplex: http://netmf.codeplex.com/workitem/2194

dimso, 2015-05-21:
Some say that bit shift operators were working in NETMF 4.1. Could anyone please suggest where to start looking for someone who would be interested in fixing this issue? In other words, where should these operators be defined?

smaillet-ms, 2015-05-21:
In the future, please leave CodePlex issue where they are - we are aware of them and will bring them into the GitHub system when they become active. Duplicating them here only adds more work to filter out duplicates etc...

The problem is more complex than a simple function, It's in the compiler itself, not the NETMF run time. For reasons unknown the compiler requires references to, at a minimum, System.Nullable(of T), which isn't available in NETMF and ultimately isn't even used by the resulting IL code!. Fixing that means modifying the compiler, Implementing all of generics support in NEMTF or some sort of hackish re-writer that can strip out the class after not being used... none of those are simple and easy.

dimso, 2015-05-21:
I have installed NETMF 4.2 (there was no support for VB in NETMF 4.1) and I have tried using bit shift operators - they were working. I am attaching a print screen as a proof.
[image]

I have also checked that System.Nullable(of T) is not supported in NETMF 4.2. So, I assume it is not required for bit shift operators to work. Could you please investigate it? Maybe there are some simple workarounds that could make bit shift operators working again?

smaillet-ms, 2015-05-22:
As I stated, I have looked into it and there is no simple workaround I could find, of course as an open source project, if you can find a solution we are happy to review and consider it for inclusion. The compiler itself is the issue, and we don't own the compiler. There is however, a very simple workaround in code. Use integer division instead of bit shifts.

miloush, 2015-05-22:
It looks the compiler team used to make reasonable effort to support language features without the generics requirement (including Linq). Perhaps you could talk to them and suggest this is welcome? Also since Roslyn is an open source project, rather than injecting IL transforms into the build process, couldn't we contribute to Roslyn so that it is capable of generating code we can use?

dimso, 2015-05-22:
Do I understand it correctly that NETMF 4.2 and 4.3 have used different compilers?

smaillet-ms, 2015-05-22:
@dimso - Not that I'm aware of no. However, the compiler may well be inferring things from runtime or framework version numbers that are different. We don't really know why the compiler is doing what it is doing, but it is in the compiler not the runtime. The build target files are all the same, and as far as I can tell so are the command line options.

dimso, 2015-10-20:
It is very sad to find out that more than half a year ago reported bug is not going to be fixed in 4.4. I have a feeling that nobody has seriously looked into this issue and that was probably because VB.NET programmers are not that important :-1:

smaillet-ms, 2015-10-20:
Feel free to offer a pull request if you can figure out what's actually going on.

dimso, 2015-10-20:
I find it interesting to read comments like yours. Someone reports a bug and in return gets a response - "Fix it yourself". I believe it is obvious that I would have already done it, if I could.

I think this bug can only be fixed by someone like you or alike, who has very good in-depth knowledge of what is going on.

The most worrying thing here is your attitude. If it is not going to change, and it doesn't look like it, this bug is never going to be addressed/fixed.

smaillet-ms, 2015-10-20:
My apologies I meant no offense. However, as discussed in this topic I'm as clueless as the rest on what the problem is. We have spent a good deal of time looking into this and don't have a solution. If somebody has one, or more importantly if someone is able to pinpoint the actual problem, I'm more than happy to hear it.

miloush, 2015-10-20:
Not sure if that is good news or not, but I have no problems compiling and debugging (in emulator) the code on the screenshot above with the Roslyn compiler. (Getting VB projects to run with that one is a bit tricky though.)