NETMF/netmf-interpreter #98: Errors when building firmware solutions

marcfrei, 2015-05-08:
I am trying to build the firmware solutions.

Starting point:

  - Microsoft Windows 8.1 Pro, Version 6.3.9600 Build 9600
  - Microsoft .NET Framework Version 4.5.51650
  - Microsoft Visual Studio Professional 2012, Version 11.0.61030.00 Update 4
  - Microsoft Visual Studio Community 2013, Version 12.0.31101.00 Update 4
  - Microsoft Visual Studio 2013 SDK
  - MDK-ARM Compiler 5.05 (build 41)

  - netmf-interpreter on branch "dev", commit https://github.com/NETMF/netmf-interpreter/commit/8b606ebfbeba07afd01a2cfbc1eb2750edd6dab7

The "Getting Started" guide on the Wiki lists Visual Studio 2013 and Visual Studio SDK as a pre-qequisite for building the SDK.

However, it appears that these packages are also required for building a firmware solution. With only Visual Studio 2012 installed, I get the following error messages:

1) Solutions "MCBSTM32E", "Template", "Template_BE", and "Windows2":

    C:\NETMF\netmf-interpreter\BuildEnv.props(66,5): error MSB4175: The task factor
    y "CodeTaskFactory" could not be loaded from the assembly "C:\Windows\Microsoft
    .NET\Framework\v4.0.30319\Microsoft.Build.Tasks.v12.0.dll". Could not load file
     or assembly 'file:///C:\Windows\Microsoft.NET\Framework\v4.0.30319\Microsoft.B
    uild.Tasks.v12.0.dll' or one of its dependencies. The system cannot find the fi
    le specified. [C:\NETMF\netmf-interpreter\Framework\Tools\BuildTasksInternal\In
    ternalBuildTasks.csproj]

2) Solution "MCBSTM32F400":

    C:\NETMF\netmf-interpreter\devicecode\Targets\Native\STM32F4\STM32F4.settings(3
    8,9): error NETMFBLD: Missing CMSIS installation at: C:\NETMF\netmf-interpreter
    \CMSIS\* [C:\NETMF\netmf-interpreter\Solutions\MCBSTM32F400\dotnetmf.proj] 

After installing Visual Studio 2013 and Visual Studio 2013 SDK, the solution Windows2 builds fine.

But building the other solutions still results in error messages:

3) Solution "MCBSTM32E":

    Processing file: scatterfile_bootloader_MDK.xml
      "C:\NETMF\netmf-interpreter\BuildOutput\THUMB2\MDK5.05\le\FLASH\release\MCBST
      M32E\bin\Tinybooter_scatterfile.txt", line 12 (column 5): Warning: L6312W: Em
      pty Execution region description for region ER_RAM_RO
    EXEC : error : L6218E: Undefined symbol ARM_Vectors (referred from STM32_intc_f
    unctions.obj). [C:\NETMF\netmf-interpreter\Solutions\MCBSTM32E\TinyBooter\TinyB
    ooter.proj]
      Not enough information to produce a SYMDEFs file.
      Not enough information to produce a FEEDBACK file.
      Not enough information to list image symbols.
      Finished: 3 information, 1 warning and 1 error messages.
    C:\NETMF\netmf-interpreter\tools\targets\Microsoft.SPOT.System.MDK.targets(313,
    5): error MSB3073: The command ""C:\Keil\ARM\ARMCC\bin\armlink.exe"  --no_keep_
    init_arrays --feedback C:\NETMF\netmf-interpreter\BuildOutput\THUMB2\MDK5.05\le
    \FLASH\release\MCBSTM32E\bin\Tinybooter_MCBSTM32E_FLASH_Release_MDK5.05.feedbac
    k  --debug --remove --unmangled --info sizes,totals,veneers,unused,libraries --
    map  --xref --symbols --callgraph --userlibpath C:\NETMF\netmf-interpreter\Buil
    dOutput\THUMB2\MDK5.05\le\FLASH\release\MCBSTM32E\lib --userlibpath C:\NETMF\ne
    tmf-interpreter\BuildOutput\THUMB2\MDK5.05\le\ANY_MEDIA\release\lib --libpath C
    :\Keil\ARM\RV31\LIB --libpath C:\Keil\ARM\ARMCC\LIB  --entry EntryPoint  --symd
    efs C:\NETMF\netmf-interpreter\BuildOutput\THUMB2\MDK5.05\le\FLASH\release\MCBS
    TM32E\bin\Tinybooter.symdefs --list C:\NETMF\netmf-interpreter\BuildOutput\THUM
    B2\MDK5.05\le\FLASH\release\MCBSTM32E\bin\Tinybooter.map --output C:\NETMF\netm
    f-interpreter\BuildOutput\THUMB2\MDK5.05\le\FLASH\release\MCBSTM32E\bin\Tinyboo
    ter.axf --scatter C:\NETMF\netmf-interpreter\BuildOutput\THUMB2\MDK5.05\le\FLAS
    H\release\MCBSTM32E\bin\Tinybooter_scatterfile.txt C:\NETMF\netmf-interpreter\B
    uildOutput\THUMB2\MDK5.05\le\FLASH\release\MCBSTM32E\obj\Solutions\MCBSTM32E\Ti
    nyBooter\Allocator.obj C:\NETMF\netmf-interpreter\BuildOutput\THUMB2\MDK5.05\le
    \FLASH\release\MCBSTM32E\obj\Solutions\MCBSTM32E\TinyBooter\TinyBooterEntry.obj
      Crypto.lib GlobalLock_hal_Cortex.lib TinyHal_Cortex.lib cpu_prestackinit_stub
    s.lib STM32_bootstrap.lib cpu_cache_stubs.lib cpu_ebiu_stubs.lib STM32_GPIO.lib
     STM32_IntC.lib cpu_mmu_stubs.lib STM32_Power.lib STM32_time.lib STM32_SPI.lib
    STM32_USART.lib STM32_USB.lib cpu_watchdog_stubs.lib system_initialization_hal_
    loader.lib Watchdog_pal_stubs.lib asyncproccall_pal.lib blockstorage_pal.lib Bu
    ttons_pal.lib COM_pal.lib DebuggerPort_SSL_config_stubs.lib i2c_pal_stubs.lib u
    sart_pal.lib usb_pal.lib config_pal_loader.lib SimpleHeap.lib SimpleHeap_config
    .lib heap_pal.lib piezo_pal_stubs.lib tinycrt_pal_loader.lib fs_pal_stubs.lib F
    S_Config_stubs.lib Ink_pal_stubs.lib Gesture_pal_stubs.lib palevent_pal_stubs.l
    ib Time_pal.lib TimeService_pal_stubs.lib backlight_hal_stubs.lib batterycharge
    r_hal_stubs.lib batterymeasurement_hal_stubs.lib lcd_hal_stubs.lib cpu_LCD_stub
    s.lib STM32_Flash.lib M25P64_Flash.lib STM32_blconfig_MCBSTM32E.lib M25P64_blco
    nfig_MCBSTM32E.lib BlockStorage_AddDevices_MCBSTM32E.lib IO_Init_MCBSTM32E.lib
    events_pal.lib sockets_pal_stubs.lib usb_pal_config_MCBSTM32E.lib C:\NETMF\netm
    f-interpreter\BuildOutput\THUMB2\MDK5.05\le\FLASH\release\MCBSTM32E\lib\tmp_Tin
    ybooter.lib" exited with code 1. [C:\NETMF\netmf-interpreter\Solutions\MCBSTM32
    E\TinyBooter\TinyBooter.proj]

3) Solutions "Template", and "Template_BE":

    EXEC : error : C9912E: No --cpu selected [C:\NETMF\netmf-interpreter\Support\Wi
    reProtocol\dotNetMF.proj]
    C:\NETMF\netmf-interpreter\tools\targets\Microsoft.SPOT.System.MDK.targets(278,
    4): error MSB3073: The command ""C:\Keil\ARM\ARMCC\bin\armcc.exe"  --cpp  --dia
    g_suppress 2874,111,161,550,C3011,C4052 --diag_suppress 66,161,230,1293 --split
    _sections  -DMDK_V3_1 -DARM_V3_1 --feedback C:\NETMF\netmf-interpreter\tools\ma
    ke\Feedback\TEMPLATE_MDK5.05.feedback -DTINYCLR_ENABLE_SOURCELEVELDEBUGGING  -D
    COMPILE_ARM  -DVERSION_MAJOR=4 -DVERSION_MINOR=4 -DVERSION_BUILD=0 -DVERSION_RE
    VISION=0 -DOEMSYSTEMINFOSTRING="\"Microsoft Copyright (C) Microsoft Corporation
    .  All rights reserved.\"" -DPLATFORM_ARM_TEMPLATE  -DTARGETLOCATION_FLASH  -DL
    ITTLE_ENDIAN     --arm  --fpu softvfp   --littleend  --no_debug --dwarf2 --no_d
    ebug_macros -O2 --inline -Otime --no_autoinline   -DNDEBUG  -IC:\NETMF\netmf-in
    terpreter\Support\WireProtocol -IC:\NETMF\netmf-interpreter\DeviceCode\include
    -IC:\NETMF\netmf-interpreter\DeviceCode\Cores\arm -IC:\NETMF\netmf-interpreter\
    Support\Include   -IC:\NETMF\netmf-interpreter\crypto\inc  -IC:\NETMF\netmf-int
    erpreter\CLR\Include -IC:\NETMF\netmf-interpreter\CLR\Libraries\CorLib -IC:\NET
    MF\netmf-interpreter\CLR\Libraries\SPOT -IC:\NETMF\netmf-interpreter\CLR\Librar
    ies\SPOT_Hardware -IC:\NETMF\netmf-interpreter\CLR\Libraries\SPOT_Graphics -IC:
    \NETMF\netmf-interpreter\CLR\Libraries\SPOT_Net -IC:\NETMF\netmf-interpreter\CL
    R\Libraries\SPOT_Net_Security -IC:\NETMF\netmf-interpreter\Solutions\Template -
    IC:\NETMF\netmf-interpreter\devicecode\Targets\Native\TEMPLATE -IC:\NETMF\netmf
    -interpreter\DeviceCode\Cores\arm -IC:\NETMF\netmf-interpreter\DeviceCode\Cores
    \arm\Include -IC:\NETMF\netmf-interpreter\DeviceCode -IC:\NETMF\netmf-interpret
    er\DeviceCode\Include -IC:\NETMF\netmf-interpreter\DeviceCode\Targets\Native\TE
    MPLATE\TEMPLATE -IC:\NETMF\netmf-interpreter\Support\Include -IC:\NETMF\netmf-i
    nterpreter\DeviceCode\include -IC:\NETMF\netmf-interpreter\CLR\include -IC:\NET
    MF\netmf-interpreter\Crypto\inc -IC:\NETMF\netmf-interpreter\BuildOutput\ARM\MD
    K5.05\le\ANY_MEDIA\release\lib -IC:\Keil\ARM\RV31\INC    -o C:\NETMF\netmf-inte
    rpreter\BuildOutput\ARM\MDK5.05\le\ANY_MEDIA\release\obj\Support\WireProtocol\C
    ompression.obj -c C:\NETMF\netmf-interpreter\Support\WireProtocol\Compression.c
    pp  " exited with code 1. [C:\NETMF\netmf-interpreter\Support\WireProtocol\dotN
    etMF.proj]

4) Solution "MCBSTM32F400":

    C:\NETMF\netmf-interpreter\devicecode\Targets\Native\STM32F4\STM32F4.settings(3
    8,9): error NETMFBLD: Missing CMSIS installation at: C:\NETMF\netmf-interpreter
    \CMSIS\* [C:\NETMF\netmf-interpreter\Solutions\MCBSTM32F400\dotnetmf.proj]

Any ideas on what could cause these problems?

Thank you very much,
Marc

smaillet-ms, 2015-05-08:
1) Yes, VS2013 is required for building the code in the source tree (not just the SDK, sorry for the confusion we'll see about updating the wiki to clarify that better)
2) MCBSTM32F400 - That error is informing you that he CMSIS libraries from ARM are needed but not found. At this time, we don't include the source in the NETMF tree (There's some legal mumbo jumbo on that point we are trying to figure out a good answer to, but as you can probably imagine, anything involving corporate attorneys is a pain! ;-) ) The folder %SPOCLIENT%\CMSIS\ should contain a readme to provide details on where to get the correct version of the CMSIS libraries. We definitely need to document that one better.
3)  Solution MCBSTM32E is not currently being maintained and we really should remove it from the GitHub repo unless someone wants to own maintaining it. (It is a "little brother" chip to the one used in the MCBSTM32F400)
4) The Template and Template_BE are not buildable as they only represent the basic templates for creating a new solution and are not a complete solution for any real hardware.

marcfrei, 2015-05-09:
Thank you very much, Steve.

After instllaing CMSIS, I also had to add the include path

    <IncludePaths Include="CMSIS\CMSIS\RTOS\RTX\INC" />

to the following two settings files

    DeviceCode\Targets\Native\STM32F4\STM32F4.settings

    DeviceCode\Targets\OS\CMSIS_RTOS\CMSIS_RTOS.settings

Now the solution MCBSTM32F400 seams to compile fine but unfortunately I get an error from armlink.exe:

    EXEC : Fatal error : L6002U: Could not open file CMSIS_RTX.lib: No such file or
     directory [C:\NETMF\netmf-interpreter\Solutions\MCBSTM32F400\TinyCLR\TinyCLR.p
    roj]
      Not enough information to produce a FEEDBACK file.
      Finished: 1 information, 0 warning, 0 error and 1 fatal error messages.
    C:\NETMF\netmf-interpreter\tools\targets\Microsoft.SPOT.System.MDK.targets(313,
    5): error MSB3073: The command ""C:\Keil\ARM\ARMCC\bin\armlink.exe"  --verbose
    --keep *(.init_array) --no_keep_init_arrays --feedback C:\NETMF\netmf-interpret
    er\BuildOutput\THUMB2FP\MDK5.05\le\FLASH\release\MCBSTM32F400\bin\tinyclr_MCBST
    M32F400_FLASH_Release_MDK5.05.feedback  --debug --remove --unmangled --info siz
    es,totals,veneers,unused,libraries --map  --xref --symbols --callgraph --userli
    bpath C:\NETMF\netmf-interpreter\BuildOutput\THUMB2FP\MDK5.05\le\FLASH\release\
    MCBSTM32F400\lib --userlibpath C:\NETMF\netmf-interpreter\BuildOutput\THUMB2FP\
    MDK5.05\le\ANY_MEDIA\release\lib --libpath C:\Keil\ARM\RV31\LIB --libpath C:\Ke
    il\ARM\ARMCC\LIB  --entry EntryPoint  --symdefs C:\NETMF\netmf-interpreter\Buil
    dOutput\THUMB2FP\MDK5.05\le\FLASH\release\MCBSTM32F400\bin\tinyclr.symdefs --li
    st C:\NETMF\netmf-interpreter\BuildOutput\THUMB2FP\MDK5.05\le\FLASH\release\MCB
    STM32F400\bin\tinyclr.map --output C:\NETMF\netmf-interpreter\BuildOutput\THUMB
    2FP\MDK5.05\le\FLASH\release\MCBSTM32F400\bin\tinyclr.axf --scatter C:\NETMF\ne
    tmf-interpreter\BuildOutput\THUMB2FP\MDK5.05\le\FLASH\release\MCBSTM32F400\bin\
    tinyclr_scatterfile.txt C:\NETMF\netmf-interpreter\BuildOutput\THUMB2FP\MDK5.05
    \le\FLASH\release\MCBSTM32F400\obj\Solutions\MCBSTM32F400\TinyCLR\tinyclr_dat.o
    bj C:\NETMF\netmf-interpreter\BuildOutput\THUMB2FP\MDK5.05\le\FLASH\release\MCB
    STM32F400\obj\Solutions\MCBSTM32F400\TinyCLR\allocator.obj C:\NETMF\netmf-inter
    preter\BuildOutput\THUMB2FP\MDK5.05\le\FLASH\release\MCBSTM32F400\obj\Solutions
    \MCBSTM32F400\TinyCLR\tinyclr.obj C:\NETMF\netmf-interpreter\BuildOutput\THUMB2
    FP\MDK5.05\le\FLASH\release\MCBSTM32F400\lib\InteropAssembliesTable.lib LargeBu
    ffer_hal_stubs.lib virtualkey_hal_stubs.lib Crypto_stub.lib GlobalLock_hal_RTOS
    .lib cpu_prestackinit_stubs.lib STM32F4_bootstrap.lib batterycharger_hal_stubs.
    lib batterymeasurement_hal_stubs.lib STM32F4_GPIO.lib STM32F4_Power.lib STM32F4
    _time.lib cpu_watchdog_stubs.lib MCBSTM32F400_initialization_hal.lib MCBSTM32F4
    00_tinyhal.lib Watchdog_pal_stubs.lib asyncproccall_pal.lib blockstorage_pal.li
    b Buttons_pal.lib config_pal.lib heap_pal.lib piezo_pal_stubs.lib SimpleHeap_st
    ubs.lib SimpleHeap_config_stubs.lib tinycrt_pal.lib fs_pal_stubs.lib Ink_pal_st
    ubs.lib Gesture_pal_stubs.lib palevent_pal.lib Time_pal.lib FS_Config_stubs.lib
     TimeService_pal_stubs.lib backlight_hal_stubs.lib lcd_hal_stubs.lib cpu_cache_
    stubs.lib cpu_LCD_stubs.lib STM32F4_Flash.lib STM32F4_SPI.lib STM32F4_I2C.lib S
    TM32F4_USART.lib STM32F4_USB.lib STM32F4_INTC.lib STM32F4_PWM.lib STM32F4_Analo
    g.lib STM32F4_DA.lib COM_pal.lib DebuggerPort_SSL_Config_stubs.lib events_pal.l
    ib usart_pal.lib usb_pal.lib i2c_pal.lib MFUpdate_PAL_stub.lib cpu_ebiu_stubs.l
    ib cpu_mmu_stubs.lib cpu_performancecounter_stubs.lib CMSIS_RTX.lib CMSIS_RTX_C
    onfig.lib timeservice_hal_stubs.lib sockets_pal_lwip_os.lib lwip_1_4_1_os_CMSIS
    _RTOS.lib SocketsDriver_PAL_LWIP_os.lib sockets_hal_sockets_lwIP_os.lib sockets
    _hal_dhcp_LWIP_os.lib sockets_hal_tcp_lwIP_os.lib sockets_hal_udp_lwIP_os.lib S
    TM32F4_ETH_lwip_os.lib Crypto_Config_MCBSTM32F400.lib openssl_pal.lib libssl_op
    enssl.lib ssl_types.lib SimpleStorage_PAL.lib PKCS11_Engine.lib OpenSSL_PKCS11_
    Crypto.lib crypto_openssl.lib Network_Config_HAL_MCBSTM32F400.lib M29W640FB_Fla
    sh.lib M29W640FB_blconfig_MCBSTM32F400.lib STM32F4_blconfig_MCBSTM32F400.lib Bl
    ockStorage_AddDevices_MCBSTM32F400.lib IO_Init_MCBSTM32F400.lib usb_pal_config_
    MCBSTM32F400.lib native_double_pal.lib C:\NETMF\netmf-interpreter\BuildOutput\T
    HUMB2FP\MDK5.05\le\FLASH\release\MCBSTM32F400\lib\tmp_tinyclr.lib" exited with
    code 1. [C:\NETMF\netmf-interpreter\Solutions\MCBSTM32F400\TinyCLR\TinyCLR.proj
    ]

Thanks again.

smaillet-ms, 2015-05-09:
You do NOT need to add any additional include paths - something else is wrong as the existing include paths are already in place as needed. Which version of the CMSIS did you get? The readme is explicit about the version (It's not the most recent version yet, as that came out only a short while ago and breaks everything as the directory structure and APIs are different. We need to deal with that before we RTM but it's not there yet.)

marcfrei, 2015-05-09:
Thanks again, Steve.

I got the correct version of CMSIS now (CMSIS-SP-00300-r4p2-00rel0.zip). With this version MCBSTM32F400 builds just fine.

Thank you very much for you help.

closed 2015-05-09 by marcfrei