NETMF/netmf-interpreter #109: Create public room for the repository on gitter.im

scout119, 2015-05-14:
I really like the quick feedback loop that gitter provides. See the dotnet/roslyn room:
https://gitter.im/dotnet/roslyn

Would be nice if you guys create public room for NETMF/netmf-interpreter.

Thank you,
Valentin

smaillet-ms, 2015-05-19:
Out of curiosity, what do you see as the benefits of using that over using the existing GitHub system. I looked at the link provided and couldn't really see any value in gitter vs. GitHub. (Not that I think GitHub is the greatest thing since sliced bread, at this point I just don't see gitter is really offering anything better)

scout119, 2015-05-20:
Gitter is an addition to GitHub. It is the GitHub's chat system with major mobile and desktop clients. Just very convenient to use. Doesn't cost anything to enable it. A good place for quick Q@A or any other live discussions. Totally optional :)

Thanks,
Valentin

zezba9000, 2015-07-22:
@smaillet-ms Its a place to chat with like minded people and ask questions.  GitHub discussions are good but its not the same thing.  With gitter we can ask question or talk about stuff we work on that may not be good for this section.

smaillet-ms, 2015-07-22:
@zezba9000 But the conversation still ends up here in GitHub anyway AFAIK, so that's why I'm unclear on the value.

zezba9000, 2015-07-22:
Because I can have coreclr open, Roslyn open ect all in one window.  You can see who is online, who read your post, who is typing (I think).  You can have lots of channels and topics open and you don't need to get emailed everytime someone messages you.  It works like IRC but keeps track of text in case of a connection break or you put your computer to sleep.  GitHub issues is like a forum not really very interactive for quick chats.  Maybe an IRC channel would be better but Gitter is just being used by all the other .NET projs is why it might be nice.

If the conversation goes on Github where (didn't know this [chat info isn't kept around forever I wouldn't think)?   Also it probably requires you to refresh your page all the time to get updates.

closed 2018-03-23 by scout119