NETMF/netmf-interpreter #115: Expand available hardware information

piwi1263, 2015-05-20:
The tool MFDeploy shows in the Device capabilities, some of them are available in the SystemInfo class in Microsoft.SPOT.Hardware. Make them all available.

smaillet-ms, 2015-05-20:
What sort's of scenarios do you have in mind for additional information?

piwi1263, 2015-05-20:
Uniquely identify the hardware by f.e. a serial number. Detect a build date
to determine whether the software will run, since sometimes the version
won't be enough. I don't know if already possible, at least I oversaw it
then but querying the CPU like STM32F411RET6, amount of flash and ram could
be of benefit during deployment scenarios.

How to set these infos during FW build and of course to query them by using
managed code afterwards.

For uniquely id a dev when using its sensor data to store in the cloud. I
could use the MAC but that's no real ID for the underlaying HW board.

Some information, like parts of the HAL information displayed (as mentioned
above) in the dev caps display using MFDeploy can't be reached through
managed code and requires native code what is not always possible.

I assume the unused properties are never filled since they can't be
approached by managed code, if at all.

lt72, 2015-07-14:
Moving to 4.5, as this is related to the HAL configuration changes