NETMF/netmf-interpreter #128: Eliminate or suppress warnings for build_sdk (VS14)

cw2, 2015-05-22:
After upgrade to VS14 (2015) the build process started from build_sdk.cmd generates a number of warnings:

#### C++ Compiler warnings
warning C4267: '=': conversion from 'size_t' to 'short', possible loss of data
warning C4838: conversion from '<type1>' to '<type2>' requires a narrowing conversion

In Debug configuration there are 500+ Microsoft.Design and Microsoft.Performance warnings.

cw2, 2015-05-22:
I volunteer to fix (in NETMF code) and suppress (in 3rd party libs, OpenSSL) C++ compiler warnings.

I am not sure what to do with Code Analysis warnings.

smaillet-ms, 2015-05-22:
Yes, the VS2015 compiler is more aggressive and picky about things and the OpenSSL code is particularly covered in legal but potentially problematic code. It's very difficult to write portable C or C++ code that doesn't trigger warnings on at least one compiler somewhere.

smaillet-ms, 2015-06-17:
Fixed with pull request #132

closed 2015-06-17 by smaillet-ms