NETMF/netmf-interpreter #155: Support-GCC branch AppEntry.s GCC assembler errors

cw2, 2015-06-07:
The AppEntry.s in Support-GCC branch (`netmf-interpreter/Solutions/MCBSTM32F400/DeviceCode/TinyHal/GNU_S/AppEntry.s`) lines 99 and 157 have incorrect comment character `;` (most likely copied from its MDK counterpart), causing assembler errors.

smaillet-ms, 2015-06-08:
Please note: a fairly significant rework of the Suppport-GCC branch was pushed up over the weekend. This was not done as a normal pull request due to numerous GIT merge issues resulting in a request that overloaded GitHub. (No browser I could find was able to render it in a way it could be pull from the GIthub site, let alone reviewed)

closed 2015-06-08 by smaillet-ms

cw2, 2015-06-09:
Sorry, but the errors are present in the latest version.

Since this is the second time a possible problem with the "weekend push" is mentioned, could you please clarify that you mean commit from 6/6, i.e. e46ff38615dcb790cf7c86620a0b2bef42f55237 and earlier? Or should be there any others?

smaillet-ms, 2015-06-10:
Keep in mind this is a dynamic development branch off of the dev branch so some churn and instability is expected. The only reason it is present in the main repo and site is to allow for further collaboration and experimentation with the GCC support as we go since we know many people have interest here. Unfortunately gcc is a bit of a beast and the documentation isn't exactly centralized in one place for a particular build of the tools so it's been a lot slower going then we hoped for. 

At the moment there are two sets of the assembly files. The solution ones are currently needed to handle the alternate startup sequence involving the normal C runtime to get to main() as the CMSIS-RTOS (and CMSIS-RTX) are intended to be used. Ultimately there's very little different between them and the plan is to alter the process so there's only one common set (e.g. the solution specific files will be deleted) At this point the networking solution doesn't even build. (Though TinyBooter and the TinyCLR_Nonet projects to build and run) so there's more work to be done.