NETMF/netmf-interpreter #160: SSL Memory Leak

grapenutz, 2015-06-10:
There was a known memory leak when using SSL in previous versions of NETMF. It doesn't appear that the issue was brought over from codeplex nor does it appear to have been resolved.

See: http://netmf.codeplex.com/workitem/2005

smaillet-ms, 2015-06-10:
Thanks for the feedback.
We are, of course, aware of all the issues posted on codeplex. We are only re-posting them here as they become actively worked on. So just because it isn't shown here on GitHub yet doesn't mean it got lost or won't be looked at. 

As always, if you have a fix please fill out a [Contribution License Agreement](https://github.com/NETMF/netmf-interpreter/wiki/Contribution%20License%20Agreement) and post a pull request.

grapenutz, 2015-06-10:
Thanks for the update. I'd love to help. I've cloned the repository and followed the steps in the getting started page and I can successfully build the NETMF solution. So that's a good start.

However I'm having trouble trying to get started with modifying the source files. I can't find a solution file which loads up all the projects. TinyCLR.sln seems to be abandoned and not maintained. There are several other solution files which do load up like the one for MFDeploy but I seem to be missing references like MFDeployEngine.

If someone was new to the NETMF community and wanted to contribute, how would that person know where to begin with just a mess of files and no solution or project to tie it all together?

I'm hoping its possible to work with the code in a tool like visual studio and you're not all just coding this in notepad and searching for references with grepWin.

smaillet-ms, 2015-06-10:
NETMF is designed to build using a number of different compiler and CPU architecture tool chains. At the time it began VS didn't support extensibility on that front for C++ projects so it's all a custom MSBUILD based system.Thus, there is no single solution for the code. (While there used to be one, it was grossly out of date and wouldn't actually build anything correctly anyway causing more confusion and trouble then it was worth so we removed it.) There are a few we've updated and set up for the build that we use internally and we'll continue to add more as we need them but a wholesale re-write of the entire build infrastructure would be needed to really support VS the way we'd all like and that's not something we can afford the time to do any time soon.

smaillet-ms, 2015-06-10:
As to what tools we use, different developers do different things depending on personal style but we usually use VS for editing and search the code from the search option in VS.

lt72, 2016-03-02:
attaching repro from GHI

[SSLLeakTest.zip](https://github.com/NETMF/netmf-interpreter/files/155038/SSLLeakTest.zip)