NETMF/netmf-interpreter #164: Wrong C# long and double formatting in GCC build with newlib-nano

cw2, 2015-06-11:
The managed application running in TinyCLR built with GCC and newlib-nano produces wrong results of `long.ToString()` and `double.ToString()`:
```
long l = 1;
Debug.Print(l.ToString());
double d = 1;
Debug.Print(d.ToString());
```
Output
```
ld
ld.0000000000000000lu
```
This is because newlib-nano does not support [C++] long-long and long-double required for native formatting routines (printf).

There are configuration options `--enable-newlib-io-long-long` and `--enable-newlib-io-long-double` that can be used to build custom version of 'regular' newlib, so it has the needed support, but I am not sure those are available for newlib-nano too.

cw2, 2015-06-23:
I have built GCC ARM Toolchain with `--enable-newlib-io-long-double`, binaries and modified scripts are available for download at

* https://cw2.github.io/downloads/gcc-arm-none-eabi-4_9-2015q1-20150306-netmf-win32.7z
* https://cw2.github.io/downloads/gcc-arm-none-eabi-4_9-2015q1-20150306-netmf-src.zip

Or in the repo https://github.com/cw2/cw2.github.io/tree/master/downloads

To build the toolchain, follow the instructions in the official "How to build" guide, extract the contents of the zip file before step "Start building the toolchain" (page 6).

It is also needed to disable the use of `nano.specs` in linker options - I'll submit pull request with the modification, so it is applied only during TinyCLR build, but right now the GCC build is severely broken, so I am waiting for the fixes to be merged into Support-GCC branch.

After that, I'll also publish firmware binaries.

smaillet-ms, 2015-06-23:
Can you submit those changes to the ARM Embedded GCC team? We'd like to keep the officially supported build of GCC to the version they distribute in binary form from the LaunchingPad site. 

What's "severely broken" in the GCC build? Which fixes are you waiting on to be merged?

cw2, 2015-06-23:
Well, I am not sure - on one side, it would be really nice to have the binaries available from the official site, on the other side, imagine what would happen if everyone wanted to have a toolchain built by ARM team with different options - without long-long, with long-long, with long-long and long-double etc. (and there are many of such options). I guess it would add significant burden to them, not only to build the binaries but also to test them and support them. As far as I understand it, their resources are rather limited.

IMVHO you are in a much better position to start a discussion with them :innocent: 

Regarding "severely broken", I was referring to wchar_t USB issue #193 - fix for which has just been merged, thanks!

smaillet-ms, 2015-06-23:
Well at the least then this should be done as an optional parameter in the build with a check to ensure that the correct version of tool or libraries is installed when the option is enabled. That way the default path remains using the officially released binaries and those hitting edge cases can decide what they want to do.

closed 2015-06-23 by smaillet-ms