NETMF/netmf-interpreter #174: STM32F4Discovery: Remove buttons driver from TinyBooter

cw2, 2015-06-16:
It seems to be unnecessary to link the Buttons driver in TinyBooter project. The [only] user button could be handled by simple 'get pin state' during startup, but even that is questionable: the same functionality, i.e. switching to bootloader mode can be achieved via MFDeploy command "Reboot and Stop" (a little bit unfortunate translation of "Connect to TinyBooter") and flashing the firmware via NETMF bootloader has to be done from MFDeploy anyway.

Otherwise, onboard ST-LINK or STM32 ROM bootloader can be used to flash both NETMF bootloader and firmware, so the user button is not needed too in this scenario.

_This also saves a few bytes, with a little bit more size optimizations, the bootloader could hopefully fit into 32 KB._

doingnz, 2015-06-16:
NetMF supports wide range of HW, not just STM32. Be careful about assumptions for other systems purely on what can or cannot be done with STM32. 

There may be legacy systems that depend on this functionality. Depending on how a system starts up, it may or may not be listening for MFDeploy. e.g. if a port is not dedicated to NetMF, but shared in some way. It is then necessary to inform the system to start up differently and begin to listen for MFDeploy in order to invoke the commands to perform the firmware updates. 

Obviously this can be achieved in several ways and need not be done via IO (button) at start up. The advantage of the early IO check is that it is simple to implement and is a way to recover access to a system when higher level software systems fail or become inaccessible.

In a secure system, the option of access to the ROM functionality to take over and control a system would usually be entirely disabled unless that access ensured that all of memory is erased before access is granted.

If you do remove Buttons from STM32, try do so in a way that makes the same functionality easily added back for those who do needed it, and conversely makes it easy to remove for those who do not want it.

smaillet-ms, 2015-06-17:
Almost completely agree with both views here. In general, there are a lot of libraries pulled in to applications that really aren't needed and can't be linker optimized out since there's an InitializeXXX call of some sort in the startup path that keeps it in. As we think through working on top of CMSIS, the upcoming mBed 3.0m or any other existing runtime we need to re-evaluate how we handle the componentization better so that component libraries can be completely removed (without neededing to plug in a stub!) when not needed.

cw2, 2015-06-17:
@doingnz  Please note the proposed change affects STM32F4DISCOVERY board only, not STM32 in general or any other platform. Basically, it means to clean up code in `...\STM32F4DISCOVERY\TinyBooter\TinyBooterEntry.cpp`, which is already commented out.

cw2, 2015-06-25:
Merged by @mortezag in pull request #208. Thanks.

closed 2015-06-25 by cw2