NETMF/netmf-interpreter #194: Branch Support-GCC/STM32F4 Discovery: wrong native checksum for Windows.Devices assembly

maiorfi, 2015-06-21:
At boot:

 Loading start at 8050f0c, end 80644dc

  ...
   
__Invalid native checksum: Windows.Devices 0xF4E46B74!=0xF71D0B80__

...
Error: 96000000

Waiting for debug commands...

cw2, 2015-06-21:
As a quick workaround, update the value in `g_CLR_AssemblyNative_Windows_Devices` structure at the end of `CLR\Libraries\Windows_Devices\windows_devices_native.cpp`, or comment out all lines that contain "Windows_Devices" in TinyCLR.proj in the Discovery solution (to exclude the assembly from firmware image).

I guess there could be a problem in one of the recent merges - to be investigated...

maiorfi, 2015-06-22:
Indeed I did it and solved, thanks. Nevertheless it's likely to be a real bug due to a mismatch between interop native and managed parts, isn't it?

smaillet-ms, 2015-06-23:
There was a bug on that and fixed in dev. The GCC-Support branch hasn't been merged with the current dev in a while. Given that there are now a few people successfully making progress on GCC it may be worth just rolling it up into the dev branch and avoiding further discrepancies. It was originally set up so that multiple people could get going while potentially needing to modify core files to compile properly with GCC. Now that we are past that stage there doesn't seem to be much of a reason to keep this in a separate branch anymore. @maiorfim, @cw2  Thoughts?

cw2, 2015-06-23:
:+1: :ok_hand: :satisfied:

maiorfi, 2015-06-24:
It would be great!

piwi1263, 2015-06-24:
You've got my big thumbs up as well !!! :+1: :+1:

smaillet-ms, 2015-06-24:
ok, sounds like consensus to me. :tongue: 
It might be a couple days before I can get to doing a merge, hopefully that will go easily as we did a fairly big rebase/merge not too long ago and I don' think there are any areas where I'll be stuck in manual merge hell...

smaillet-ms, 2015-06-24:
Well that was easier than I thought! Fortunately the previous large rebase/merge worked out all the kinks and only a couple small conflicts existed. So it all went in. Woohoo! :joy:

cw2, 2015-06-24:
:clap:

closed 2015-06-25 by smaillet-ms