NETMF/netmf-interpreter #195: Microsoft.SPOT.Time's Environment.TickCount replacement?

maiorfi, 2015-06-22:
Hi.

In order to test brand new tcp/ip stack on NETMF 4.4, I have recently slightly modified and rebuilt Paolo Patierno's amazing M2Mqtt library.

The first (and only) glitch I found was that such library makes use of Environment.TickCount property, that seems stubbed on reference STM32F4 porting.

Is there anything wrong in using DateTime.Now.Ticks (provided that one manages different unit of measure, i.e. ticks instead of milliseconds)?

Is there a better replacement? Will Environment.TickCount ever be implemented through interop?

Thanks!

ppatierno, 2015-06-22:
Hi,

in the past I avoided to use DateTime.Now.Ticks for performance reasons (after some experimental tests working on .Net Compact Framework on Windows CE). From there I decided to use Environment.TickCount on .Net Micro Framework too and of course for my M2Mqtt library.

As Lorenzo said, will Environment.TickCount ever be implemented ? Will it be replace by another method ?

Or as last solution, I'll have to use DateTime.Now.Ticks .... I hope without loss performance ?

Thanks!

cw2, 2015-06-22:
Well, I don't see a problem with implementing Environment.TickCount - as long as it is not too complicated or adds too much overhead. Personally, I would like to keep the STM32F4DISCOVERY solution as simple as possible, but functional and useable in real world scenarios. I'll look into it later today... but of course contributions are welcome!

However, I have to note that I am neither member of the Microsoft team nor official maintainer of the Discovery solution.

cw2, 2015-06-22:
Do you need the whole TimeService implementation? Because if not, I'd propose that  Environment.TickCount is moved from TimeService.cs to corlib.

ppatierno, 2015-06-22:
I didn't try it on 4.4 but do you mean that the TimeService will be removed from 4.4 version ?

cw2, 2015-06-22:
No, not removed. In the current implementation, Environment.TickCount is included in TimeService.cs - so in order to make it work, I'd need to add the whole TimeService (SNTP) project into solution. I am not sure it is good idea, especially on a board that does not have networking...

Instead, I'd like to move Environment.TickCount implementation from TimeService.cs to some corlib file and implement its (one line function call) native counterpart there. Then, we would have working Environment.TickCount and the TimeService remains as it is.

maiorfi, 2015-06-22:
Since all in all we're talking about getting current machine time, I followed interop code behind DateTime.Now, getting to STM32F4_time_functions.cpp, where Oberon's gurus defined main (and only) backing function as follows:

`
UINT64 __section(SectionForFlashOperations) HAL_Time_CurrentTicks()
{
    UINT32 t2, t3; // cascaded timers
    do {
        t3 = TIM_16->CNT;
        t2 = TIM_32->CNT;
    } while (t3 != TIM_16->CNT); // asure consistent values
    return t2 | (UINT64)t3 << 32;
}
`

So, despite I'm quite surprised to see that such function returns after a non deterministic time, waiting for next tick to come, I think that (int)(DateTime.Now.Ticks/10000) is a good replacement for Environment.TickCount, (at least in managed code).

smaillet-ms, 2015-06-25:
While I didn't write that code, and unfortunately there weren't any comments explaining the reasoning, I can imagine it is one of consistency of timestamps for data logging. By waiting until the time transitions you ensure you are consistent with timestamps in all cases, thus reducing the impact of jitter on continuous sampling over time computations. This has a disadvantage of stalling the CPU for periods of time on systems that don't need low jitter time stamping though.

cpfister, 2015-06-29:
The 32 bit timer counts along with the 16 bit sysclock timer. The latter is incremented for every reset of the 32 bit counter. Together they form a 48 bit counter, which is what NETMF needs for sys ticks.

The loop prevents inconsistencies that could occur if the reset occurs exactly between the two read operations. In this case, the time would be off by 2^32. The loop makes sure to repeat the read in this (and only this) case. Thus the loop is executed at most two times, and normally only once.