NETMF/netmf-interpreter #196: Branch Support-GCC|STMF4DISCOVERY

piwi1263, 2015-06-22:
The Discovery board has MCU STM32F407VGT6, correct ?

If yes, shouldn't the RAM be 192 KB and not 128 KB as is found in several scatter files in your STM32F4DISCOVERY solution or am I misreading that ??

http://www.st.com/web/catalog/tools/FM116/SC959/SS1532/LN1848/PF252419

cw2, 2015-06-23:
Those 192 KB of RAM consists of two separate regions: 128 KB SRAM + 64 KB CCM. In the current Discovery board configuration, whole CCM is dedicated to stack, leaving most of the SRAM available for the heap.

piwi1263, 2015-06-23:
Thanks for the clarification, I did misread that. FWIW: I'm currently trying to use your STM32F4DISCOVERY solution as a bases and am trying to get that to work with the NUCLEOF411 ...

First try after 3 hours of searching and replacing etc. in notepad got a build fail, but will try again of course. Any idea whether the full compile report is stored ?

cw2, 2015-06-23:
You can use msbuild command line parameters to specify logging output, I usually run something like
```
cd Solutions\STM32F4DISCOVERY
msbuild /m /v:m /flp1:verbosity=detailed;LogFile=build.log /flp2:LogFile=build_err.log;ErrorsOnly
```

I am planning on getting NUCLEO board mid-July, but most likely one of the lower-end variants.

piwi1263, 2015-06-23:
Thanks, I'll do it that way.

BTW, just saw the 407DISCO board arrived, will be fun later when I get to
it. I'll see how the 4.4 does it on the board.

smaillet-ms, 2015-06-23:
@cw2 A Note on using msbuild /m on NETMF projects: It can cause troubles with build failures in unexpected ways. Due to the legacy history of the build infrastructure not all of the projects have dependencies defined in a way that the multi-threaded msbuild support can see them. This can cause the build to try and generate binaries out of order. If it's working for you, great! It's definitely a build time improvement, if you switch machines, get a faster hard drive or just decide to overclock your machine you could end up with failures and need to remove the /m.

cw2, 2015-06-23:
@smaillet-ms Thanks for the explanation. IIRC I encountered some problems with parallel build in previous versions of NETMF, unfortunately I don't remember which one. I think from 4.3 the build process is much less prone to multi-threading related issues, I am not aware of any problems with using /m. But, I agree there could be some issues after hardware change or when some corner case is hit.

smaillet-ms, 2015-06-23:
I've not seen issues with solution builds. On the SDK build we had to disable the /m and even had to force msbuild to unload itself when done instead of remaining lingering for a few minutes to speed up the next run. I found radically different behaviors on my desktop, our build sever and on my home machine. All due to the dependencies being built in a non-deterministic order. Fixing all of that is on the list of things to do but not for the v4.4 release cycle - we don't want to upset something so central at a late stage of the game.

closed 2015-06-25 by smaillet-ms