NETMF/netmf-interpreter #209: Help appreciated on build issue

robvanschelven, 2015-06-24:
I get build errors on both STM32F4Discovery as well as on MCBSTM32F400
Build_SDK and windows2 build proceed without errors

Anyone a suggestion? Thanks in advance!

[image]

smaillet-ms, 2015-06-24:
I'm not clear on the message as I'm not fluent in that particular language :grin: However I can see from the command window title you have used setenv_vs - which won't work for a solution build targeting the STM32F400. You need to use a new command prompt window using setenv_gcc or setenv_mdk depending on the tool set you are using. mixing setenv_xxx calls in the same command window is not generally a supported scenario.

robvanschelven, 2015-06-25:
PLATFORM_INDEPENDENT_LIB_DIR wasn't defined because of a failure i made typing the compiler version :( Now i was able to build MCBSTM32F400 however STM32F4DISCOVERY builds with 1 error

[image]

cw2, 2015-06-25:
Please update to the latest version, that (#204) was fixed in pull request #205.

robvanschelven, 2015-06-25:
Thanks @cw2 that solved the problem!

closed 2015-06-25 by smaillet-ms