NETMF/netmf-interpreter #211: STM32F4DISCOVERY: Add managed hardware helpers

cw2, 2015-06-25:
Sorry about the title, I mean what is implemented in `MCBSTM32F400\ManagedCode\Hardware\CPU.cs`. A few points open for discussion

* What namespace?  Microsoft.SPOT.Hardware.STM32F4DISCOVERY or rather something like Community.NETMF.Hardware.STM32F4DISCOVERY ?
* Follow .NET naming guidelines more closely. IMHO `Pins.None`, `Pins.A8` is better than `Pins.GPIO_NONE`, resp. `Pins.GPIO_A8`
* How to handle LEDs and User button? Two constants, i.e. Pins.PD13 and Pins.LED3, or one Pins.PD13_LED3?

smaillet-ms, 2015-06-25:
I'd recommend avoiding the SPOT namespace as we are trying to phase that out. I agree  that Pins.None and Pins.A8, etc.. fit better into the normal .NET naming conventions.  For overlapping pins I'd use two separate names with the same value so that you can be explicit about intent to help in comprehension, readability and reviews. (e.g. The intention is clearly to use an LED, which implies an output, but the code uses an InputPort... Hmmm something smells funny here... )

kodfilemon, 2015-06-27:
Namespace: using Solutions.Stm32F4Discovery?
All pins: Pins.PA0 .. Pins.PE15 is ok.
In my code I repeated some pins: LedPins.Blue = Pins.PD15, ButtonPins.User, FreePins.PA1 but maybe nested types like Pins.Led.Green, Pins.Button.User?

cpfister, 2015-06-29:
One of the nice things about the .NET ecosystem compared to Java is the (relative) consistency of names in APIs, and of API design in general. Now that we are about to get rid of SPOT namespaces, NETMF has a chance to get closer to this .NET quality as well. No more all-caps identifiers (even for abbreviations longer than two characters), no more underscores, etc.

I like the brevity of Solutions.Stm32F4Discovery, but there may be multiple solutions for the same board, supporting a different (sub)set of features, and the solutions may come from different sources. Thus Community.NETMF.Hardware.Stm32F4Discovery seems more appropriate to me. (All caps for STM32F4DISCOVERY would be allowed because it can be considered a brand, but fortunately this is not required. In contrast, Pins.LED3 is not allowed according to the design guidelines https://msdn.microsoft.com/en-us/library/ms229027(v=vs.110).aspx)