NETMF/netmf-interpreter #212: VB debug static variable examination causes crash

rockybooth, 2015-06-25:
I have been debugging a large VB program and having lots of debug issues and decided to try to find the cause. With the following not so long program, I can crash Visual Studio (2012 or 2013) every time by simply waiting for the stop, then using the cursor to see the present value of I. Using the immediate window to find the value will also crash VB. 

Local variables declared inside the sub can be explored without issue but variables passed to a sub or function also cannot be examined in that sub or function. 

Similar code in C# using Static instead of Private works just peachy.

This happens with the emulator, so this is not a GHI issue, and none of their dlls are loaded.

Option Explicit On
Option Strict On

Imports Microsoft.SPOT
Imports Microsoft.VisualBasic.CompilerServices

Namespace MFConsoleApplication1

    Public Module Module1
        Private I As Integer = 1

        Sub Main()

            Debug.Print(Resources.GetString(Resources.StringResources.String1))
            Stop
            Debug.Print(I.ToString)
        End Sub

    End Module

End Namespace

 
Presumably the issue is in Microsoft.VisualBasic.dll
This occurs running 
Microsoft.Spot.Native 4.3.1
mscorlib 4.3.1
Microsoft.Visualbasic 1.0.0.0 (v4.0.30319)

I hope I am posting in the right place.  Also see https://www.ghielectronics.com/community/forum/topic?id=19600&page=1#msg193443
where this is discussed

smaillet-ms, 2015-07-07:
Thanks for reporting this, we'll investigate and post what we find here. Yes, this is the right place to post bugs and other discussions on the .NET Micro Framework.

rockybooth, 2016-04-07:
Can anyone comment on when this might be resolved or if there are workarounds?
Thanks

smaillet-ms, 2016-04-10:
Other than not using statics we don't have any workarounds yet. We're working on core runtime and porting issues first (such as a memory leak in OpenSSL)

We're more than happy to have someone look into this, if someone is interested feel free to claim this one here.

martincalsyn, 2018-09-10:
I have a fix for this which I will submit as a PR once I clean it up a bit.  The changes are all in the CorDebug component. Basically CorDebugClass has to return an instance of CorDebugGenericType in the CorDebugClass implementation of ICorDebugClass2.GetParameterizedType.  And CorDebugGenericType needs an implementation for GetStaticFieldValue.  Those two changes stop the crashing and will return proper hover and watch values for VB globals.

martincalsyn, 2018-09-10:
Should work for both, as the change is in the Visual Studio debugging code which is in the MicroFramework Visual Studio Extension (vsix), which is common to 4.3 and 4.4.  I have built and tested it for VS2017.  I haven't built vsix packages for earlier versions (yet).