NETMF/netmf-interpreter #217: Separate ITM tracing from serial

cw2, 2015-06-27:
In the current implementation, ITM is handled as a virtual serial port - while it should be possible to workaround the issue #10 by using port number that does not affect the existing USART numbering scheme (such as 254) (?), I would suggest separating ITM from serial transport and make it independent.

STM32F4DISCOVERY tinybooter would not need USART support at all (`DEBUG_TEXT_PORT` is already set to `ITM0`).

smaillet-ms, 2015-06-29:
The reason it was done as a virtual serial port was to minimize the number of places where there is a switch based on the port type (such as in COM re-director) while I'm not a fan of that particular anti-pattern (switch on type) there's a lot of it in NETMF since using virtuals in some of the older tool chains used for the earlier days of NETMF caused significant code bloat. At the time, getting something up and going quickly was more important than solving the problem more broadly. Using an arbitrarily high port number has other issues given how the various re-director and other APIs use the Max port count as indices into arrays or tables and assumes that all ports are sequentially numbered. Ultimately the best solution is to ditch the redirector switching on type to using function pointer dispatching (or virtuals) to fully isolate the consumer from implementation. This has the added advantage of removing another layer of stubs for all the port types a system doesn't support but has to be there to satisfy the linking. Ideally this could be done where only common PAL/HAL code is changed and existing HAL drivers remain unchanged.

cw2, 2015-06-30:
That would be great.

I've found another problem related to ITM - it does not work in Tinybooter. Respectively, it now only works when all four ports (DEBUG_TEXT_PORT, STDIO, DEBUGGER_PORT and MESSAGING_PORT) are set to the same value and it has to be either USBx or COMx.

In Tinybooter (Initialization\tinyhal.cpp) the communication initialization function CPU_InitializeCommunication() is never called (conditional compilation), so the initialization of STDIO depends on its type - if it is set to ITM0, it never happens, if it is set to USBx or COMx, it is initialized when either debugger or messaging does it, whichever is called first.

smaillet-ms, 2015-06-30:
@cw2 Can you file a new bug/issue on that, as it's broader than just the ITM implementation issues.

closed 2015-07-10 by smaillet-ms