NETMF/netmf-interpreter #218: STDIO set to ITM0 is not initialized in TinyBooter

cw2, 2015-06-30:
_Split from #217_

I've found another problem related to ITM - it does not work in Tinybooter. Respectively, it now only works when all four ports (DEBUG_TEXT_PORT, STDIO, DEBUGGER_PORT and MESSAGING_PORT) are set to the same value and it has to be either USBx or COMx.

In Tinybooter (Initialization\tinyhal.cpp) the communication initialization function CPU_InitializeCommunication() is never called (conditional compilation), so the initialization of STDIO depends on its type - if it is set to ITM0, it never happens, if it is set to USBx or COMx, it is initialized when either debugger or messaging does it, whichever is called first.

smaillet-ms, 2015-06-30:
As discussed the issue is technically unrelated to ITM. It has to do with how the startup and transport initialization and startup work in tinybooter vs. the other projects. It's not clear why that was done differently and if there is any size/perf benefits to short cutting that over keeping things consistent and standard. My current recommendation on this is to modify the tinybooter start up so it is consistent and calls CPU_InitializeCommunication().

smaillet-ms, 2015-07-01:
I'll fix this at the same time as I fix the STM32 UART code. (in progress now)

PR #223 (2015-07-07): Fix itm virtual port on STM32F4xx
fixes #218, and #10
closed 2015-07-07 by smaillet-ms