NETMF/netmf-interpreter #221: RTC initialization

josesimoes, 2015-07-01:
Hi,

I'm looking into RTC and I need some clarifications, please.
In the current implementation I believe RTC_Initialize() (@ STM32F4_RTC\STM32F4_RTC_functions.cpp) is being called in Time_Initialize() (@ STM32F4_Time\STM32F4_time_functions.cpp).

Despite that I also find in line #279 @ STM32F4_Bootstrap\STM32F4_bootstrap.cpp the following

#ifdef STM32F4_Enable_RTC
    STM32F4_RTC_Initialize(); // enable RTC
#endif

But STM32F4_Enable_RTC doesn't show anywhere else.

I would like to understand better what is the strategy for RTC initialization and configuration.
Is it supposed to be implemented  @ STM32F4_RTC_functions.cpp like it is now?
Maybe the STM32F4_RTC_Initialize() code block was from a previous implementation or it something that's planned but not implemented...

Appreciate your insight on this.
Thanks!

josesimoes, 2015-07-07:
hello... anyone...please...
I just need some clarifications on how this was planned (assuming someone gave it some thought sometime ago).
If there are no plans/reasoning about it I'll go ahead and make some proposals for discussion.

smaillet-ms, 2015-07-07:
We didn't write this code it came from the Oberon/CSA port so we'll need to investigate, same as you. Proposals for change are always welcome.

josesimoes, 2015-07-08:
Thanks Steve. That helps. :)
I'll get back to this thread with a proposal for the RTC implementation.

cpfister, 2015-07-09:
The code shown was used for an experimental version of the RTC driver and is no longer valid.

The actual driver is integrated into NETMF through a special version of the standard PAL Time driver function wrapper (time_function.cpp).

The RTC version is called time_function_rtc.cpp and is located in STM32F4\DeviceCode\STM32F4_RTC\Time_Pal_Rtc.

josesimoes, 2015-07-09:
Thanks Cuno!

Untitled86, 2016-09-26:
My RTC doesn't seem to be initializing.  Should there be a reference to ..\Time_Pal_Rtc\dotNetMF.proj in my TinyCLR.Proj?  I see the time_functions_rtc.cpp file and its project but I don't see anything referencing it.

josesimoes, 2016-09-27:
@Untitled86 take a look at #359. There is a bit more that just pulling in that project to actually get the RTC to work and have the register value in/out the C# <code>DateTime</code>.