NETMF/netmf-interpreter #225: Feature: Named Captures in Regular Expressions

NameOfTheDragon, 2015-07-06:
Please support named captures in regular expressions.

As a test case, this expression:  

    const string CommandRegex = @"<(?<Address>(\w\d)),(?<Transaction>\d+),(?<Verb>[A-Za-z]\w+)(=((?<NumericPayload>\d+)|(?<TextPayload>.+)))?>";
    const RegexOptions options = RegexOptions.Compiled | RegexOptions.CultureInvariant | RegexOptions.ExplicitCapture | RegexOptions.Singleline;
    static readonly Regex Parser = new Regex(CommandRegex, options);

Should produce named captures named "Address", "Transaction", "Verb", "NumericPayload" and "TextPayload" with the last two being optional and mutually exclusive. This expression compiles correctly and works in "desktop" applications, but in .NET Micro Framework it throws 
<pre>An unhandled exception of type 'System.Text.RegularExpressions.RegexpSyntaxException' occurred in System.Text.RegularExpressions.dll
Additional information: Syntax error: Missing operand to closure
</pre>

This is a lamentable omission because Named Captures are one of the features that make regular expressions more bearable and lead to much cleaner code. Named captures also have great tooling support in products such as ReSharper.

The following test code should be able to use the above expression to match the example inputs shown in the comment block and retrieve the named captures (again this works correctly on "desktop" apps and even compiles in .netMF, since the Match class has the Groups string indexer):

            /*
             * Commands are understood in one of three basic formats:
             * <F1,234,Nickname=Fred>   -- non-numeric payload
             * <F5,999,MoveTo=1000>     -- numeric payload
             * <F3,0,Center>            -- verb only (no payload)
             * If the RegEx match was successful, then we must have one of these three possibilities.
             * The address, transaction ID and command verb must always be present so we get those first.
             */
            var deviceAddress = matches.Groups["Address"].Value;
            var transaction = int.Parse(matches.Groups["Transaction"].Value);
            var verb = matches.Groups["Verb"].Value;

            // Now we must decide which of the three basic forms we are dealing with.
            // For the verb-only case, we simply set some defaults before checking for a payload.
            var position = Command.NoPosition;
            var data = string.Empty;

            if (matches.Groups["Position"].Success)
                {
                // We have a numeric payload
                position = int.Parse(matches.Groups["NumericPayload"].Value);
                }
            if (matches.Groups["Data"].Success)
                {
                // We have a non-numeric payload
                data = matches.Groups["TextPayload"].Value;
                }
            var source = matches.Value;
            /*
             * Commands are understood in one of three basic formats:
             * <F1,234,Nickname=Fred>   -- non-numeric payload
             * <F5,999,MoveTo=1000>     -- numeric payload
             * <F3,0,Center>            -- verb only (no payload)
             * If the RegEx match was successful, then we must have one of these three possibilities.
             * The address, transaction ID and command verb must always be present so we get those first.
             */
            var deviceAddress = matches.Groups["Address"].Value;
            var transaction = int.Parse(matches.Groups["Transaction"].Value);
            var verb = matches.Groups["Verb"].Value;

            // Now we must decide which of the three basic forms we are dealing with.
            // For the verb-only case, we simply set some defaults before checking for a payload.
            var position = Command.NoPosition;
            var data = string.Empty;

            if (matches.Groups["Position"].Success)
                {
                // We have a numeric payload
                position = int.Parse(matches.Groups["NumericPayload"].Value);
                }
            if (matches.Groups["Data"].Success)
                {
                // We have a non-numeric payload
                data = matches.Groups["TextPayload"].Value;
                }
            var source = matches.Value;

lt72, 2015-07-14:
moved to 4.5

NameOfTheDragon, 2015-08-14:
A cautionary note: I have found the capture group code in the current version (4.3) is very broken. It 'sort of' works for some input strings, but suffers from arbitrary truncation of the matched strings and throws `OutOfRangeException` when trying to access groups that should be present. For example, it will report that there are 8 capture groups, but group[0] will be truncated to only a a partial match and perhaps groups [4..7] will throw an exception. This code needs some serious TLC.

Please refer to my question on StackOverflow to see how much hair I have pulled out: http://stackoverflow.com/questions/31502055/using-net-micro-framework-why-does-my-regular-expression-match-fail