NETMF/netmf-interpreter #229: Cannot find any entrypoint!

cpfister, 2015-07-08:
This is probably an instance of various build and deploy issues, here one file overwriting another one during the build process:

http://netmf.codeplex.com/workitem/2311

smaillet-ms, 2015-07-08:
Can you provide more details on your step 12 - Deploy? In particular what precise commands, keyboard actions, toolbar buttons or menus are involved here? (e.g. when you say deploy do you mean F5? or literally the deploy only command, which if you don't have binaries should probably either fail or auto rebuild the needed binaries) It will be tough to reproduce and fix this one without knowing the specific pathways through the VS project system.

cpfister, 2015-07-08:
I don't know Etan's commands at the time, but I'm pretty sure that when I experienced the same thing, I just used F5.

In my case, I had a solution with several projects in it. When I built/deployed the first project, it may have worked. But then I changed the Default Project to the next one, and then this problem reliably occurred. For every one of the projects that I switched to. Both for Mountaineer boards and Netduino 2 boards.

Etan1, 2015-07-08:
Usually, I simply hit the Start button in the toolbar (With mouse, not F5). I never use "Only Deploy".

smaillet-ms, 2015-07-08:
OK, so there's more than one project in the solution? Do they depend on one another? Are they using project references or just referencing the other's binary outputs. (If the latter then changing to project references would get the automatic dependency build to occur - referencing binaries only won't )

Etan1, 2015-07-08:
Step 1 describes the solution as "Create new Project > NETMF application project".

So, no dependencies etc.

The bug occurs in quite a few scenarios, the one I described back then was a very simple reproducible test case.

One thing that may be unusual is that I'm developing in a VMware Windows 8.1 VM on OS X.

cpfister, 2015-07-08:
I think at the time I had a far more complicated solution, but not used a virtual machine.

smaillet-ms, 2015-07-13:
#### Text from original issue:
When deploying a clean solution for the first time, the error "Cannot find any entry point." is shown and deployment fails.

Steps to reproduce:
 1. Create a NETMF application project. 
 2. Deploy - it should work. 
 3. Close VS. 
 4. Remove the following: 
 5. bin folders 
 6. obj folders 
 7. DOTNETMF_FS_EMULATION folders 
 8. OnBoardFlash.dat 
 9. Trace.txt 
 10. *.suo 
 11. Open the solution created in number 1 
 12. Deploy

Expected results:
- Deployment succeeds.

Actual results:
- Deployment fails with a similar to the error message at the bottom of this report. 
- When deploying a second time, deployment succeeds.

Workarounds:
- Deploy twice.

Version information:
◾NETMF 4.3 QFE1 
◾Both VS2012 and VS2013 affected.
```
Found debugger!

Create TS.

 Loading start at 80705f0, end 8099d30

   Assembly: mscorlib (4.3.1.0)     Assembly: Microsoft.SPOT.Native (4.3.1.0)     Assembly: Microsoft.SPOT.Hardware (4.3.1.0)  
   Assembly: Microsoft.SPOT.Hardware.SerialPort (4.3.1.0)     Assembly: Microsoft.SPOT.Hardware.Usb (4.3.1.0)  
   Assembly: Microsoft.SPOT.Hardware.PWM (4.3.1.0)     [...] Loading Deployment Assemblies.

Resolving.

The debugging target runtime is loading the application assemblies and starting execution.
Ready.

Cannot find any entrypoint!

Done.

Waiting for debug commands...

'Microsoft.SPOT.Debugger.CorDebug.12.dll' (Managed): Loaded 'C:\Program Files (x86)\Microsoft .NET Micro Framework\v4.3\Assemblies\le\mscorlib.dll', Symbols loaded.
'Microsoft.SPOT.Debugger.CorDebug.12.dll' (Managed): Loaded 'C:\Program Files (x86)\Microsoft .NET Micro Framework\v4.3\Assemblies\le\Microsoft.SPOT.Native.dll', Symbols loaded.
[...]
The program '[1] Micro Framework application: Managed' has exited with code 0 (0x0).
```