NETMF/netmf-interpreter #232: Installing too many updates with MFUpdate seems to fill Config region

cpfister, 2015-07-08:
http://netmf.codeplex.com/workitem/2236

smaillet-ms, 2015-07-13:
#### Text from original issue:
MFUpdate uses the Config region to indicate whether an update has been installed. During testing we found that installing firmware updates too often can result in a condition where no new updates can be installed anymore. Restoring the Config region in MFDeploy solves the problem, until it occurs again.

 As the configuration region is intended for long term storage, we recommend modifying MFUpdate (or more precisely, the MicroBooter) to use a mechanism that doesn't depend on the Config region.

##### UPDATE 29-Jul-2014
The MFUpdate Software Update Framework for NETMF was designed in a way that it can recover from power failures. However, there is at least one situation where recovery might fail:

 Each call to MFUpdate.Install writes an entry to the config region, telling the bootloader about a pending update in the update region. Since the config region is realized in flash, old entries cannot be deleted but just overridden. Hence, eventually the config region flash is full, and has to be rewritten (garbage collection). The implemented procedure copies the config region to RAM, deletes the flash used by the config region, and rewrites still valid entries to the config region. If a power failure occurs shortly after the config flash was deleted, some (if not all) config information will be lost. Furthermore, the copy operation might fail if there is not enough memory on the heap available.

 Obviously, this behavior is not optimal for an embedded system.

###### Possible solutions:
 1) For software updates it would be sufficient, if the pending update flag would be written to a reserved flash location in the software update region (instead of the config region). It does not solve the general config region garbage collection issue described above, if the config region is dynamically updated by other parts of the system.
 2) Instead of copying the config region to RAM, an extra flash block would enable a safe garbage collection of config entries.

lt72, 2015-07-14:
The way the config sector is handled needs to be revised to achieve both better transactional behavior and not filling up the dirty bits array... we will need to postpone this to next release to create the right design.

cpfister, 2015-07-14:
Sure. But please decouple the MFUpdate mechanism from the config section.

Robust remote firmware updates (MFUpdate) is a required feature in nearly any commercial project, and this never-quite-completed mechanism urgently needs more love. We have addressed some of its problems on the firmware side, but not everything, and the brittle tool side remains an issue. We are right now suffering through such problems, and the resulting delays directly affect a customer project.

lt72, 2015-07-27:
Understood