NETMF/netmf-interpreter #234: Wrong exception for disposed socket

cpfister, 2015-07-08:
Using a disposed socket causes a SocketException, instead of an ObjectDisposed exception.

ColinA-MSFT, 2015-07-22:
It appears this works with the methods I've tried.  A scan of the entire class suggests all methods have a dispose check.  Can you provide a code sample that reproduces the issue?

cpfister, 2015-07-23:
It's a long time ago, but if I remember correctly there were indeed dispose checks. That wasn't the issue though. The issue was that they didn't always throw the correct type of exception as per the specification of the full .NET framework (SocketException where ObjectDisposedException would have been correct).

ColinA-MSFT, 2015-07-23:
It looks like that's been corrected.  All dispose checks are throwing ObjectDisposedException.  I'm going to close this out for now.  However, please don't hesitate to reopen this issue if you observe this again on 4.4.

closed 2015-07-23 by ColinA-MSFT