NETMF/netmf-interpreter #235: FileStream.Flush() does not flush data immediately

cpfister, 2015-07-08:
http://netmf.codeplex.com/workitem/2149

smaillet-ms, 2015-07-13:
#### Text from original issue:
According to MSDN, a stream's Flush method should write buffered data to the underlying device, implying that this should happen without major delay.

http://msdn.microsoft.com/en-us/library/system.io.stream.flush(v=vs.102).aspx

 However, it can take up to a minute until this actually happens on NETMF devices. The usual workaround is to call the NETMF-specific method VolumeInfo.FlushAll, e.g.:

 new Microsoft.SPOT.IO.VolumeInfo("\SD").FlushAll();

 but this should not be necessary.

 See also  https://www.ghielectronics.com/community/forum/topic?id=13546&page=1#msg138957.

ColinA-MSFT, 2015-08-13:
After discussing with the team, we've decided we'll address this in the next release.  This requires some extra foundational work in order to reproduce.