NETMF/netmf-interpreter #237: ResXFileCodeGenerator not handling namespace collisions

cpfister, 2015-07-08:
http://netmf.codeplex.com/workitem/2329

smaillet-ms, 2015-07-09:
text of original issue on codeplex:
If a custom namespace containing "System" as one part is defined (e.g. Foo.System) in referenced code of a project that contains a *.resx embedded resource file, the generated code looks like follows:
    internal partial class FooResources
    {
        private static System.Resources.ResourceManager manager;
        internal static System.Resources.ResourceManager ResourceManager
        {
            get
            {
                if ((FooResources.manager == null))
                {
                    FooResources.manager = new System.Resources.ResourceManager("FooResources", typeof(FooResources).Assembly);
                }
                return FooResources.manager;
            }
        }
    }
The System.Resources namespace won't be find due to namespace collisions with the other referenced source files.

 Adjusting the code above to use fully-qualified namespaces fixes the issue.
    internal partial class FooResources
    {
        private static global::System.Resources.ResourceManager manager;
        internal static global::System.Resources.ResourceManager ResourceManager
        {
            get
            {
                if ((FooResources.manager == null))
                {
                    FooResources.manager = new global::System.Resources.ResourceManager("FooResources", typeof(FooResources).Assembly);
                }
                return FooResources.manager;
            }
        }
    }

smaillet-ms, 2015-07-13:
This is due to the ResxFileCodeGenerator task being overloaded by NETMF to handle the unique nature of NETMF resources. (most of the code in question is located in `%SPOCLIENT%\Framework\Tools\BuildTasks\GenerateResource.cs`) It will need to be updated to usethe global:: qualifier added to the language after that code was originally written.