NETMF/netmf-interpreter #238: OutputPath property does not support absolute paths

cpfister, 2015-07-08:
http://netmf.codeplex.com/workitem/2289

smaillet-ms, 2015-07-08:
This is due to several parts of the build using msbuild properties like this: $(ProjectDir)$(OutDir), which assumes that OutDir is always a relative path from the Project directory.

lt72, 2015-08-04:
This is too risky to fix at this stage, we will move to next release 4.5, hopefully by year end

cpfister, 2015-08-05:
The more general context of this issue: we wanted to have a Visual Studio solution setup where we can support several platforms (e.g. NETMF 4.3, NETMF 4.4 and the full desktop framework) in a way that portable source files only exist once (so we don't have to keep several copies in sync), but platform-dependent source files can exist in several variants. If I remember correctly, this issue here was one of those that we found when trying to create such an environment.

smaillet-ms, 2015-08-05:
@cpfister Thanks for the additional information. Unfortunately, as you can probably see by the recalled/closed pull requests I've made on this it is not a simple fix. While the simple fix resolves the build, it completely breaks debug and deploy for reasons presently unknown. Digging into and changing that part of the system at this stage is rather high risk. We'll keep looking into this but it isn't expected we'll get to a solution in the v4.4 release.

cpfister, 2015-08-05:
Sure, no problem. I just wanted to give a bit more context than in the original post.

Thanks and best regards

Cuno