NETMF/netmf-interpreter #254: Cannot create the Presdk, build error

FrederikWJohansen, 2015-07-23:
Hi, I have tried to build this project, but i have not have any succes with this one yet. I have follow the setup instruction and try to run the buildsdk.cmd, but i get this error evry time???

---- ERROR ----
C:\netmf-interpreter\netmf-interpreter-dev>build_sdk.cmd
Compiler: VS 14
Building PreSDK ...
Microsoft (R) Build Engine version 14.0.23107.0
Copyright (C) Microsoft Corporation. All rights reserved.

C:\netmf-interpreter\netmf-interpreter-dev\tools\scripts\Preprocess.js(29, 1) WshShell.Exec: The system cannot find the file specified.

CSC : error CS2001: Source file 'C:\netmf-interpreter\netmf-interpreter-dev\BuildOutput\public\Release\Server\dll\Microsoft.SPOT.WiX_assemblyinfo.txt' could not be found. [C:\netmf-interpreter\netmf-
interpreter-dev\Framework\Tools\WiXLib\WixLibSpotBuild.csproj]
C:\netmf-interpreter\netmf-interpreter-dev\tools\scripts\Preprocess.js(29, 1) WshShell.Exec: The system cannot find the file specified.

CSC : error CS2001: Source file 'C:\netmf-interpreter\netmf-interpreter-dev\BuildOutput\public\Release\Server\dll\Microsoft.SPOT.AutomatedBuild.BuildSigner_assemblyinfo.txt' could not be found. [C:\n
etmf-interpreter\netmf-interpreter-dev\Framework\Tools\BuildTasksInternal\BuildSigner\BuildSignerSpotBuild.csproj]
.....
--- END OF ERROR ---

I have no clue how to fix this????

smaillet-ms, 2015-07-23:
It looks like the system can't find the standard windows scripting host executable.
A few questions:

1. What OS version are you running on?
2. If you open a standard windows command prompt and run Wscript /? what happens?
3. What build of VS2015 do you have installed?

FrederikWJohansen, 2015-07-24:
Hi @smaillet-ms 
I am using windows 8.1 and VS 2015 Community edition

When I call Wscript /? it shows how to use Wscript (a dialog shows up with list of parameters to use, the help)

Regards
Frederik

smaillet-ms, 2015-07-24:
VS2015 build number (RC? RTM?)

FrederikWJohansen, 2015-07-24:
Hi @smaillet-ms 

Microsoft Visual Studio Community 2015
Version 14.0.23107.0 D14REL
Microsoft .NET Framework
Version 4.6.00081

Installed Version: Community

So answer must be RTM :-)

FrederikWJohansen, 2015-07-27:
HI @smaillet-ms 
I can now build the SDK's you have to activate the VS SDK inside VS 2015 in the RTM version!! (only the RC have an external SDK download) see: https://msdn.microsoft.com/en-us/library/bb166441%28v=vs.140%29.aspx

smaillet-ms, 2015-07-27:
Great news! I'll see about adding that information to the Wiki.

closed 2015-07-27 by smaillet-ms

deviousasti, 2016-04-19:
This happens on VS installs where C++ language features are left out.
> WshShell.Exec: The system cannot find the file specified.
is actually complaining about not being about to find the C/C++ linker.
It goes away when you modify the VS install.

Perhaps, an `fso.FileExists("cl.exe")` could print a more descriptive error.

smaillet-ms, 2016-04-22:
Please feel free to provide a pull request with your suggested fix for review.