NETMF/netmf-interpreter #255: Is STM32F429I-Discovery supported?

FrederikWJohansen, 2015-07-27:
Hi I have a STM32F429I-Discovery with a small touch screen, will this board be supported? I know I can download a version from ST (version 4.3.0) but there is no Porting Kit, and i would like to add some interop functions to an advance gfx lib i have for this board (written i pure c++).

cw2, 2015-07-27:
I cannot speak about the official support, but I am working on STM32F429IDISCOVERY solution for NETMF 4.4. The barebone version (no LCD at the moment) will be available in a few days - initially in my repo.

piwi1263, 2015-07-27:
HI CW2,

you can use the repo https://github.com/veccsolutions/Vecc.Netduino.Drivers.Ili9341 for a start.

You have to adopt the SPI pins to the SPI5 channel, and it is working on the 429 Disco via SPI. See here for further details https://www.ghielectronics.com/community/forum/topic?id=19593&page=5#msg195861

And yes, I'd be very interested to use 4.4. Tried it myself but failed miserably ... 

Cheers,

Peter.

FrederikWJohansen, 2015-07-27:
@cw2 awesome, look forward to try it out, if you share it :-)!

Becafuel, 2016-02-23:
@cw2 : do you have any news about this port or did I miss it ?

cw2, 2016-02-23:
I've just updated the repo, the solution is available in [cw2-stm32f429idiscovery](https://github.com/cw2/netmf-interpreter/tree/dev-cw2-stm32f429idiscovery) branch. But, that is version from July 2015, I have not worked on it since then and unfortunately I don't remember its state - whether it at least boots or not. Sorry.

smaillet-ms, 2016-02-23:
If you have a working version that at least boots with TinyBooter we'd be willing to accept it into the repository so that others can use/contribute to it.

Becafuel, 2016-02-25:
@cw2 : thank you very much. I will try to put my hands on it as soon as I can !

cw2, 2016-02-26:
Well, you are welcome. Although I would be a little bit skeptical about it, due to the reasons mentioned above. I'll too have look at it, hopefully this weekend - if I can get it working, I'll rebase the branch onto the latest dev and make a PR...

In case I will not be able to make it or you beat me to it, feel free to make a PR yourself :)

cw2, 2016-02-28:
I have checked the STM32F429 solution and it does not work. However, with about three fixes and memory layout settings from STM32F4DISCO I was able to get it boot with TinyBooter.

Becafuel, 2016-02-29:
Except for a small syntax error in your source, it is indeed booting with Tinybooter. But then it get stuck somewhere in the ClrStartup() method and the device does not show in Windows :(

From what I have seen so far, it is near the Debugger_WaitForCommands() method (although not 100% sure yet). But since I do not have anything to debug the code, it is not easy for me to go further, given that I do not know what to search for and where to search...

cw2, 2016-02-29:
Yes, you are right, it does not receive any commands because the USB connection does not work. To fix it, please follow the steps:

- Delete extra parenthesis in IO_Init.cpp (syntax error),
- Add `#define STM32F4_USB_HS 1` in platform_selector.h to fix the USB connection,
- Change `#define SRAM1_MEMORY_Size` value to `0x00020000`, because the external RAM is not yet enabled.

The board should then successfully report as an USB device and TinyBooter commands in MFDeploy should work.

The next steps are to extend memory layout and block range configuration, enable the external RAM and verify/fix peripheral pins. After it is confirmed working with GCC build, I'll submit a PR...

Becafuel, 2016-02-29:
The extra parenthesis was the easiest fix to find out :)

Now, I can say that it is "almost" working... Weird behaviour, however : the device does not show up in MFDeploy 4.4 although it appears in the Windows peripherals. What is weird is that the device appears in MFDeploy 4.3.1 !

Of course, it does not appear either in VS2015.

Here is the link of a screenshot showing what I describe : http://mikrobusnet.org/ftp/fw/ScreenShot_090.JPG
I will investigate and come back later if I find something interesting.

Anyway, thank you very much for your help.

cw2, 2016-02-29:
Hm, this problem has been already discussed several times (#376, #357 etc.) and I think it is some kind of USB driver caching issue caused by attaching multiple .NET boards that [ab]use the same VID/PID. You have "STM32 .NET Test" device name, while it should be "STM32F429IDISCOVERY" (see usb_config.cpp).

Unless you've specifically set that name (in MFDeploy, Configuration\USB Name), it seems the USB driver uses some old information and that is probably the reason MFDeploy does not see it.

I would suggest uninstalling the USB driver, perhaps even deleting device keys in the registry and if that does not help, then changing PID in usb_config.cpp, or at least "friend name" string descriptor value.

[image]