NETMF/netmf-interpreter #256: Compiling for STM32F411 problem

luca-saggese, 2015-07-27:
Hello all!
i was trying to compile the netmf for nucleo f411 (STM32F411RE), i've modified the platform_selector.h with all the info from datasheet.
i've builded all with gcc 4.9 and got thinibooter.hex and flash, but when i install the thinybooter the device is not recognized anymore, the only solution working i've found is precompiled package available at:
http://ingenuitymicro.com/downloads.aspx

the problem is that the configuration of the hardware is messed up and only one serial and one spi is working, plus i would like to try version 4.4.

where is something i could try to debug it?

here is my platform_selector.h:
`
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
// This file is part of the Microsoft .NET Micro Framework Porting Kit Code Samples and is unsupported. 
// Copyright (c) Microsoft Open Technologies, Inc. All rights reserved.
// 
// Licensed under the Apache License, Version 2.0 (the "License"); you may not use these files except in compliance with the License.
// You may obtain a copy of the License at:
// 
// http://www.apache.org/licenses/LICENSE-2.0
// 
// Unless required by applicable law or agreed to in writing, software distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied. See the License for the specific language governing
// permissions and limitations under the License.
// 
// Based on the Implementation for (STM32F4) by Oberon microsystems, Inc.
//
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

#ifndef _PLATFORM_STM32F411RE_SELECTOR_H_
#define _PLATFORM_STM32F411RE_SELECTOR_H_

/////////////////////////////////////////////////////////
//
// processor and features
//

#if defined(PLATFORM_ARM_STM32F411RE)
#define HAL_SYSTEM_NAME "STM32F411RE"

#define PLATFORM_ARM_STM32F4 1 // STM32F411 cpu
#define USB_ALLOW_CONFIGURATION_OVERRIDE 1

//
// processor and features
//
/////////////////////////////////////////////////////////

/////////////////////////////////////////////////////////
//
// constants
//

#define GPIO_PORTA 0
#define GPIO_PORTB 1
#define GPIO_PORTC 2
#define GPIO_PORTD 3
#define GPIO_PORTE 4
// The remaining ports are not broken out - except PH0 and PH1,
// which are deliberately omitted to keep the range continuous.

#define PORT_PIN(port,pin) ( ( (int)port) * 16 + ( pin ) )

// System clock
#define SYSTEM_CLOCK_HZ                  108000000   // 100 MHz
#define SYSTEM_CYCLE_CLOCK_HZ            108000000   // 100 MHz
#define SYSTEM_APB1_CLOCK_HZ             27000000   //  25 MHz
#define SYSTEM_APB2_CLOCK_HZ             54000000   //  50 MHz

#define SYSTEM_CRYSTAL_CLOCK_HZ           8000000   // 8 MHz external clock

#define SUPPLY_VOLTAGE_MV                    3300   // 3.3V supply

#define CLOCK_COMMON_FACTOR               1000000   // GCD(SYSTEM_CLOCK_HZ, 1M)

#define SLOW_CLOCKS_PER_SECOND            1000000   // 1 MHz
#define SLOW_CLOCKS_TEN_MHZ_GCD           1000000   // GCD(SLOW_CLOCKS_PER_SECOND, 10M)
#define SLOW_CLOCKS_MILLISECOND_GCD          1000   // GCD(SLOW_CLOCKS_PER_SECOND, 1k)

#define FLASH_MEMORY_Base               0x08000000
#define FLASH_MEMORY_Size               0x00080000 //512KB
#define SRAM1_MEMORY_Base               0x20000000
#define SRAM1_MEMORY_Size               0x00020000

#define TXPROTECTRESISTOR               RESISTOR_DISABLED
#define RXPROTECTRESISTOR               RESISTOR_DISABLED
#define CTSPROTECTRESISTOR              RESISTOR_DISABLED
#define RTSPROTECTRESISTOR              RESISTOR_DISABLED

#define TOTAL_GPIO_PORT                 (GPIO_PORTC + 1)
#define TOTAL_GPIO_PINS                 (TOTAL_GPIO_PORT*16)

#define INSTRUMENTATION_H_GPIO_PIN      GPIO_PIN_NONE

#define TOTAL_USART_PORT                3 // ITM0 + 6 physical UARTS

#define USART_DEFAULT_PORT              COM1
#define USART_DEFAULT_BAUDRATE          115200

#define DEBUG_TEXT_PORT                 ITM0
#define STDIO                           USB1
#define DEBUGGER_PORT                   USB1
#define MESSAGING_PORT                  USB1

#define TOTAL_USB_CONTROLLER            1  // Silicon has 2, but only one supported in this port at this time...
#define USB_MAX_QUEUES                  4  // 4 endpoints (EP0 + 3)

// System Timer Configuration
#define STM32F4_32B_TIMER 2
#define STM32F4_16B_TIMER 3

// Pin Configuration
#define STM32F4_ADC 3
#define STM32F4_AD_CHANNELS {0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15}

#define STM32F4_PWM_TIMER {4,4,4,4}
#define STM32F4_PWM_CHNL  {0,1,2,3}
#define STM32F4_PWM_PINS  {38,39,40,41} // PC6-PC9

#define STM32F4_SPI_SCLK_PINS {5, 29, 42} // PA5, PB13, PC10
#define STM32F4_SPI_MISO_PINS {6, 30, 43} // PA6, PB14, PC11
#define STM32F4_SPI_MOSI_PINS {7, 31, 44} // PA7, PB15, PC12

#define STM32F4_I2C_PORT     1
#define STM32F4_I2C_SCL_PIN  PORT_PIN( GPIO_PORTB, 8 ) // PB6
#define STM32F4_I2C_SDA_PIN  PORT_PIN( GPIO_PORTB, 9 ) // PB9

#define STM32F4_UART_RXD_PINS {23, 3, 39} // PB7, PA3, PC7
#define STM32F4_UART_TXD_PINS {22, 2, 38} // PB6, PA2, PC6
#define STM32F4_UART_CTS_PINS {(BYTE)GPIO_PIN_NONE, 0, (BYTE)GPIO_PIN_NONE} // GPIO_PIN_NONE, PA0, GPIO_PIN_NONE
#define STM32F4_UART_RTS_PINS {(BYTE)GPIO_PIN_NONE, 1, (BYTE)GPIO_PIN_NONE} // GPIO_PIN_NONE, PA1, GPIO_PIN_NONE

// User LEDs
#define LED3 PORT_PIN(GPIO_PORTD, 13) // PD.13 (orange)
#define LED4 PORT_PIN(GPIO_PORTD, 12) // PD.12 (green)
#define LED5 PORT_PIN(GPIO_PORTD, 14) // PD.14 (red)
#define LED6 PORT_PIN(GPIO_PORTD, 15) // PD.15 (blue)

// TinyBooter entry using GPIO
#define TINYBOOTER_ENTRY_GPIO_PIN       PORT_PIN(GPIO_PORTA, 0) // 'User' button
#define TINYBOOTER_ENTRY_GPIO_STATE     TRUE                    // Active high
#define TINYBOOTER_ENTRY_GPIO_RESISTOR  RESISTOR_DISABLED       // No internal resistor, there is external pull-down (R39)

//
// constants
/////////////////////////////////////////////////////////

#include <processor_selector.h>

#endif // PLATFORM_ARM_STM32F411RE

#endif // _PLATFORM_STM32F411RE_SELECTOR_H_

`

cw2, 2015-07-27:
Well, it is not enough to modify just platform_selector.h - you have to update also scatterfiles, blockrange definition etc. to match the microcontroller memory size and layout.

Hopefully, I'll have Nucleo-F411RE solution finished in a few days.

luca-saggese, 2015-07-27:
Hi cw2, thanks for the comment, do you have any hint on what to modify?
i've see the scatter file for gcc and update the memory size, do you mind share your progress on it?

thanks
Luca

luca-saggese, 2015-07-28:
i've modified scatterfile, putting the CCM memory in ram (as F411 doesn't have it) and reviewed memory map with the right numbers of sectors i think i've done everything... but still not working.
i've released all the changes in a fork:
https://github.com/luca-saggese/netmf-interpreter/tree/dev/Solutions/STM32F411

could someone help me in complete it?

thanks
Luca

cw2, 2015-07-28:
I have looked briefly at your files and there are numerous issues - in platform_selector.h alone, you have wrong flash memory size, invalid clock settings (there is no external crystal oscillator on the Nucleo board (at least not on mine), you need to use HSI or MCO from ST-LINK), wrong USART configuration, unless you've manually connected USB via some external circuitry the USB1 transports cannot work etc. The scatterfiles also don't seem to be correct; flash memory layout is also unchanged (from STM32F4DISCOVERY) in blockrange...

Please be patient, I have TinyBooter alive, the rest will be hopefully finished soon...

luca-saggese, 2015-07-28:
you are totally right, for some reason i've updated github with the wrong files :(
as for the clock, i though it should use the st-link one (8mhz)
for the usb1 i'm using PA12 and PA11 that does work with a precompiled thinybooter from ingnuitymicro, is it wrong?

i don't want to seems inpatient just trying to figure out it by myself

piwi1263, 2015-07-28:
@cw2: follow this thread how to activate an seperate clock to the nucleof411 and be able to forget about the st-link part, it is not even needed anymore and can be cut off if you like. Via this way you are in need of a single USB connection.

Thread describing the details: https://www.ghielectronics.com/community/forum/topic?id=18762&page=2#msg188043

cw2, 2015-07-28:
@piwi1263 Thanks for the link, a lot of information there :+1: After the first initial test with USART, I now have the USB transport working - although with a little bit cruder hand-made USB cable "adapter" :flushed: 

Unfortunately, I cannot cut off ST-LINK part of the board, because on my C-03 revision the crystal (X3) is not populated, so STM32F411 micro uses 8 MHz clock from ST-LINK. I have not yet tried the internal clock source (HSI), perhaps later - we'll see whether it is precise enough for reliable USB...

I could solder in 8 MHz crystal and missing resistors, but at the moment I have to focus on the software and 'stock' hardware - there will be enough confusion with those 3 revisions already.

cw2, 2015-07-29:
I have just published the initial version of STM32F411NUCLEO solution in my repo https://github.com/cw2/netmf-interpreter, branch [dev-cw2-stm32f411nucleo](https://github.com/cw2/netmf-interpreter/tree/dev-cw2-stm32f411nucleo).

It's in `CW2_STM32F411NUCLEO` folder, `CW2_` prefix indicates non-official variant (i.e. with modifications made by me).

Also, please note the branch is not 'stable', I'll rebase it when necessary, it will be merged into `dev-cw2` and then deleted.

A few more notes:
* It's still very early version, work in progress, a lot of FIXME-s,
* MDK only (GCC coming very soon),
* USB deployment, an appropriate cable has to be connected to Morpho pins,
* LED2 works, other peripherals are untested.
* TinyBooter turns LED2 on, pressing [blue] user button during startup/reset enters the bootloader mode.

Any feedback is welcome :)

piwi1263, 2015-07-30:
@cw2: thx, but I can do only GCC. So, I'll wait a bit plus can only fb asof next wednesday, I'm off-site until than ...

PeterKenyon, 2015-08-03:
sounds interesting keep up the good work

luca-saggese, 2015-08-05:
Wohoooo thanks CW2!
i've successfully port your scatter file to gcc and compiled the firmware, it's tested and working on my nucleo:
here are the compiled flash and your directory with updated gcc scatter files:
https://www.dropbox.com/s/13mk8b41zeqc3hj/v4.4.rar?dl=0

i was not able to make a pull request... grrr i don't want to learn how to use git!!!

thaks again

cw2, 2015-08-05:
Nice! I silently updated my dev-cw2-stm32f411nucleo branch a few days ago with working GCC build, but there are a few things that need more work. Also, the [tinyclr] firmware compiled with standard newlib (no nano) just barely fits, so I might need to modify the memory layout (swap code and deployment regions).

I would like to make a few changes through the main repository back to my dev branches, before I publish custom solutions - things like HSE bypass option, fixes for STM32F4xx symbol defines, fix for CCM init which does not exist on F411 etc.

piwi1263, 2016-08-06:
@CW2 - Although this thread is more than a year old, just wanted to let you know that your port for the 411 is after a few modifications for GCC 5.4.1 directly building and deploying to a NUCLEO-F411RE board without the ST-Link and using an external x-tal etc.

cw2, 2016-08-06:
:innocent:

piwi1263, 2016-08-06:
... hm ... was a little too much to expect to port it in a first try to the STM32F412ZG Nucleo board ...
Compile went OK, Building Tinybooter OK, Deployment OK but than no reaction from/to usb not even detected so I reckon MCU code is not running ... seems I have to have a closer look at the scattering things ...

piwi1263, 2016-08-08:
Well, I should have maybe opt for the release build, than an immediate success would have made my day. I got no idea why the debug build is failing mentioning the ER_FLASH doesn't fit into LR_FLASH. Especially on the 412 board with the extra RAM/FLASH, I guess the scatter and block defs are not at an optimum. But there is a start.