NETMF/netmf-interpreter #263: Build SDK

SytechDesigns, 2015-07-30:
Can anyone give details on building the SDK. i.e what needs to be installed. Currently I have a test pc with win10 release, vs2015 pro, the vs2015 sdk is installed - but surely you need wix? How do we install wix for vs2015? running build_sdk currently produces many many errors -

SytechDesigns, 2015-07-30:
ok - a problem seems to be the build cant find a Microsoft.SPOT.Wix.dll any where - in mf4.3 this used to be in the tools\build directory - but this does not exist in 4.4.
this is a new pc and just has win 10, vs2015 installed and the mf 4.4 directory - nothing else
What else do I need to install so the wix dll is found?

smaillet-ms, 2015-07-30:
Please see the [Getting Started](https://github.com/NETMF/netmf-interpreter/wiki/Getting%20Started) Wiki article. It provides step by step instructions. In particular getting the required wix binaries is part of the tools listed in step number 4.

SytechDesigns, 2015-07-30:
Steve, sorry I missed the setup instructions - embarrassing as I am always complaining about the lack of documentation!  Thanks for  answering so quickly

SytechDesigns, 2015-07-31:
Steve, I am still having issues with sdk build. The following is a dump of the start my console window

C:\git_repo\netmf-interpreter>build_sdk
Compiler: VS 14
Building PreSDK ...
Microsoft (R) Build Engine version 14.0.23107.0
Copyright (C) Microsoft Corporation. All rights reserved.

  C:\git_repo\netmf-interpreter\tools\scripts\Preprocess.js(29, 1) WshShell.Exe
  c: The system cannot find the file specified.

CSC : error CS2001: Source file 'C:\git_repo\netmf-interpreter\BuildOutput\publ
ic\Release\Server\dll\Microsoft.SPOT.WiX_assemblyinfo.txt' could not be found.
[C:\git_repo\netmf-interpreter\Framework\Tools\WiXLib\WixLibSpotBuild.csproj]
  C:\git_repo\netmf-interpreter\tools\scripts\Preprocess.js(29, 1) WshShell.Exe
  c: The system cannot find the file specified.

CSC : error CS2001: Source file 'C:\git_repo\netmf-interpreter\BuildOutput\publ
ic\Release\Server\dll\Microsoft.SPOT.AutomatedBuild.BuildSigner_assemblyinfo.tx
t' could not be found. [C:\git_repo\netmf-interpreter\Framework\Tools\BuildTask
sInternal\BuildSigner\BuildSignerSpotBuild.csproj]
C:\Program Files\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets(1819,
5): warning MSB3245: Could not resolve this reference. Could not locate the ass
embly "Microsoft.SPOT.WiX". Check to make sure the assembly exists on disk. If
this reference is required by your code, you may get compilation errors. [C:\gi
t_repo\netmf-interpreter\Framework\Tools\BuildTasksInternal\InternalBuildTasks.
csproj]
C:\Program Files\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets(1819,
5): warning MSB3245: Could not resolve this reference. Could not locate the ass
embly "Microsoft.SPOT.AutomatedBuild.BuildSigner". Check to make sure the assem
bly exists on disk. If this reference is required by your code, you may get com
pilation errors. [C:\git_repo\netmf-interpreter\Framework\Tools\BuildTasksInter
nal\InternalBuildTasks.csproj]
  C:\git_repo\netmf-interpreter\tools\scripts\Preprocess.js(29, 1) WshShell.Exe
  c: The system cannot find the file specified.

 This looks similar to the closed issue #254 
I have tried on my Win 10 machine, and also I have installed vs2015 ( release) on my win 7 machine, both have the sdk installed - result is the same on both machines

[image]

 VS2015 is the release version  (14.0.23107.0 D14REL)
(Enterprise on the win 7 machine and pro on the win 10 machine)
The build tools.zip  has been extracted and added the x86 folder etc. to the original tools folder, 

I get similar type results if I try and build 'build.dirproj'

Like the #254 issue, if I type wscript /? in the console, it comes back with all the command usage options.

SytechDesigns, 2015-08-01:
problem is with release version of VS2015 and SDK. Works with RC versions. Raised as a new issue #265

smaillet-ms, 2015-08-03:
We've been scratching heads around here on this one as we couldn't reproduce the problem at all. Until we did a complete clean update of our daily build server to use the RTM version of VS2015 and then it hit. But not on any of our dev machines.... After looking into the details more, it turns out that, the VS installer does not install C++ support by default, at least for the community edition of VS anyway. So you need to go back to the installer (or Programs and features and select 'change') to modify the installation. Then under the languages option in the installer be sure to select C++. That should get you back up and running