NETMF/netmf-interpreter #265: Build sdk /tools fails with VS2015 Release

SytechDesigns, 2015-08-01:
Steve, opening this as a new issue from issue #263.

Build SDK fails with VS2015 Pro and Enterprise Release versions, with 'internal' SDK install ( not tried express). If you uninstall VS2015 release and replace with VS2015 RC and separate SDK, then with same code base Build_SDK works.
With release version you get the following type of errors -
Compiler: VS 14
 Building PreSDK ...
 Microsoft (R) Build Engine version 14.0.23107.0
 Copyright (C) Microsoft Corporation. All rights reserved.

C:\git_repo\netmf-interpreter\tools\scripts\Preprocess.js(29, 1) WshShell.Exe
 c: The system cannot find the file specified.

CSC : error CS2001: Source file 'C:\git_repo\netmf-interpreter\BuildOutput\publ
 ic\Release\Server\dll\Microsoft.SPOT.WiX_assemblyinfo.txt' could not be found.
 [C:\git_repo\netmf-interpreter\Framework\Tools\WiXLib\WixLibSpotBuild.csproj]
 C:\git_repo\netmf-interpreter\tools\scripts\Preprocess.js(29, 1) WshShell.Exe
 c: The system cannot find the file specified.

CSC : error CS2001: Source file 'C:\git_repo\netmf-interpreter\BuildOutput\publ
 ic\Release\Server\dll\Microsoft.SPOT.AutomatedBuild.BuildSigner_assemblyinfo.tx
 t' could not be found. [C:\git_repo\netmf-interpreter\Framework\Tools\BuildTask
 sInternal\BuildSigner\BuildSignerSpotBuild.csproj]

VS2015 is the release version (14.0.23107.0 D14REL) - with matching sdk

cw2, 2015-08-03:
I guess it's `cl.exe` that cannot be found, most likely the PATH is not set correctly [on a clean machine with only VS 2015 and .NET Framework 4.6].

SytechDesigns, 2015-08-03:
On the same 'clean' machine, if I uninstall vs2015 release and then install vs2015 RC and separate SDK, then build_sdk builds just fine. Problem occurs when you use the release version of VS2015.
I get the same result on my Win7 machine.

cw2, 2015-08-03:
What type of VS2015 RTM installation did you select? Typical?

Because Visual C++ was removed from Typical installation (*), you have to select Custom and then check "Common Tools for Visual C++ 2015" in order to install them.

(*) [Setup Changes in Visual Studio 2015 Affecting C++ Developers](http://blogs.msdn.com/b/vcblog/archive/2015/07/24/setup-changes-in-visual-studio-2015-affecting-c-developers.aspx)

smaillet-ms, 2015-08-03:
We've been scratching heads around here on this one as we couldn't reproduce the problem at all. Until we did a complete clean update of our daily build server to use the RTM version of VS2015 and then it hit. But not on any of our dev machines.... After looking into the details more, it turns out that, as @cw2 points out, the VS installer does not install C++ support by default, at least for the community edition of VS anyway. So you need to go back to the installer (or Programs and features and select 'change') to modify the installation. Then under the languages option in the installer be sure to select C++. That should get you back up and running

SytechDesigns, 2015-08-04:
I had to build customer boards today, I will try replacing VS2015RC with the release version tomorrow, and make sure that C++ support is also installed and retry

SytechDesigns, 2015-08-05:
I have only had time to test on a Win7 (x32)  machine - but VS2015 release now builds sdk - if you make sure C++ tools installed. 
However the SDK.MSI does not get built get the following log error

Microsoft (R) Build Engine version 14.0.23107.0
Copyright (C) Microsoft Corporation. All rights reserved.

  Product.wxs
C:\git_repo\netmf-interpreter\Setup\Include\mergecrt.wxs(4): error CNDL0235: Un
defined preprocessor function '$(env.CommonProgramFiles(x86))'. [C:\git_repo\ne
tmf-interpreter\setup\ProductSDK\Product.wixproj]
C:\git_repo\netmf-interpreter\tools\Targets\Microsoft.SPOT.SDK.Targets(665,5):
error MSB3073: The command "C:\git_repo\tools\x86\WiX\tools_3_5_1315_0\candle.e
xe -nologo -IC:\git_repo\netmf-interpreter\Setup\Include -wx -out C:\git_repo\n
etmf-interpreter\BuildOutput\public\Release\Server\obj\WiX\Modules\Product\ -dB
uildFlavor=Release -dBuildTreeServer=C:\git_repo\netmf-interpreter\BuildOutput\
public\Release\Server -dBuildTreeClient=C:\git_repo\netmf-interpreter\BuildOutp
ut\public\Release\Client -dBuildTreeClientTest=C:\git_repo\netmf-interpreter\Bu
ildOutput\public\Release\test\client -dBuildTreeServerTest=C:\git_repo\netmf-in
terpreter\BuildOutput\public\Release\test\server -dWixSpoClient=C:\git_repo\net
mf-interpreter -dProdVerMajor=4 -dProdVerMinor=4 -dProdVerBuild=0 -dProdVerRevi
sion=0 -dProdReleaseName="(Simon)" -dProdVer=4.4.0.0 -dProdVerMajorMinor=4.4 -d
MSBuildProjectDirectory="C:\git_repo\netmf-interpreter\setup\ProductSDK" -dMSBu
ildProjectFile=Product.wixproj -dMSBuildProjectExtension=.wixproj -dMSBuildProj
ectFullPath="C:\git_repo\netmf-interpreter\setup\ProductSDK\Product.wixproj" -d
MSBuildProjectName=Product -dMSBuildBinPath="C:\Program Files\MSBuild\14.0\bin"
 -dMSBuildProjectDefaultTargets=TinyCLR_Build -dMSBuildExtensionsPath="C:\Progr
am Files\MSBuild" -dDesktopTargetFrameworkVersion=2.0.50727 -dVisualStudioInteg
rationDir="VisualStudioIntegration" -dINCLUDEDOCS=  Product.wxs" exited with co
de 235. [C:\git_repo\netmf-interpreter\setup\ProductSDK\Product.wixproj]
Building VSIX packages ...
Microsoft (R) Build Engine version 14.0.23107.0
Copyright (C) Microsoft Corporation. All rights reserved.

Problem is first line error -
error CNDL0235: Un
defined preprocessor function '$(env.CommonProgramFiles(x86))'. [C:\git_repo

When you setenv_vs on a X32 machine, you do not get an environment variable for 'CommonProgramFiles(x86). (As the x64 type progfiles dir does not exist)

I have the sdk msm built on my Win10 machine - with VS2015 RC - I assume that this msm will work on a x32 machine? Haven't had time to test yet.

Is their any need to modify the wix project to be able to build on x32 and x64 machines? Not really urgent if the sdk msm built on a X64 machine installs on an x32 machine, and if in mf4.4 release the sdk msm is downloadable.

Did a test build of the Discovery project and tinybooter and tinyclr build just fine. So all required tools on buildsdk have worked.

Hopefully this makes sense, its the end of a very long day....

smaillet-ms, 2015-08-05:
We haven't tested building the SDK on anything but an 64bit system in a very long time. This looks like it might be a bug in the build infrastructure somewhere. Something seems to be assuming the build is running on a 64bit system. You don't need a 64 bit machine to build the MSI. It should install fine on 32 or 64 bit machines when built from a 64 bit system.

The final release will include fully built and signed binaries of the MSI and VSIX packages and we will publish them to the VS Extension gallery so they are accessible directly within VS. Signing files is a major pain in a large corporation so we didn't produce a set of signed binaries for the pre-released versions of 4.4.

kmshim, 2015-12-27:
I found that there is path error when I build the SDK on Windows 7 32bit OS.
env.CommonProgramFiles(x86) of mergecrt.wxs is not available. so I think that if  try to build  SDK on windows 7 32bit OS,  _env.CommonProgramFiles(x86)_ should be replaced with _env.CommonProgramFiles_.
<?xml version="1.0"?>
<Include>
    <DirectoryRef Id="TARGETDIR">
        <Merge Id="CRT140" Language="0" SourceFile="$(env.CommonProgramFiles(x86))\Merge Modules\Microsoft_VC140_CRT_x86.msm" DiskId="1" />
        
        
        <?if $(var.BuildFlavor) = "Debug"?>
        <Merge Id="Debug CRT140" Language="0" SourceFile="$(env.CommonProgramFiles(x86))\Merge Modules\Microsoft_VC140_DebugCRT_x86.msm" DiskId="1" />
        
        <?endif?>
    </DirectoryRef>
</Include>

kmshim, 2016-02-02:
Hi, @smaillet-ms 
I propose that mergecrt.wxs could be modified to be built by the condition of processor architecture as following.

```xml
<?if $(env.PROCESSOR_ARCHITECTURE) = "x86" ?>
    <Merge Id="CRT140" Language="0" SourceFile="$(env.CommonProgramFiles)\Merge Modules\Microsoft_VC140_CRT_x86.msm" DiskId="1" />        
    <!-- Debug modules are not redistributable, but are useful during development and early testing -->
   <?if $(var.BuildFlavor) = "Debug"?>
      	<Merge Id="Debug CRT140" Language="0" SourceFile="$(env.CommonProgramFiles)\Merge Modules\Microsoft_VC140_DebugCRT_x86.msm" DiskId="1" />
       	<!-- Merge Id="Debug CRT Policy" Language="0" SourceFile="$(env.CommonProgramFiles(x86))\Merge Modules\policy_9_0_Microsoft_VC90_DebugCRT_x86.msm" DiskId="1" /-->
   <?endif?>
<?else?>
    <Merge Id="CRT140" Language="0" SourceFile="$(env.CommonProgramFiles(x86))\Merge Modules\Microsoft_VC140_CRT_x86.msm" DiskId="1" />        
    <!-- Debug modules are not redistributable, but are useful during development and early testing -->
    <?if $(var.BuildFlavor) = "Debug"?>
        <Merge Id="Debug CRT140" Language="0" SourceFile="$(env.CommonProgramFiles(x86))\Merge Modules\Microsoft_VC140_DebugCRT_x86.msm" DiskId="1" />
         <!-- Merge Id="Debug CRT Policy" Language="0" SourceFile="$(env.CommonProgramFiles(x86))\Merge Modules\policy_9_0_Microsoft_VC90_DebugCRT_x86.msm" DiskId="1" /-->
    <?endif?>
<?endif?>
```