NETMF/netmf-interpreter #266: XML comments

dimso, 2015-08-02:
Can anyone explain why XML comments are not provided along with NETMF assemblies?
For example, AnalogInput class (https://github.com/NETMF/netmf-interpreter/blob/dev/Framework/Core/Native_Hardware/AnalogInput.cs) has some XML comments, but we won't see them with IntelliSense (there are no XML files generated).

If that could be fixed, I would like to suggest to add XML comments for all the other classes, as most of them do no have any comments. I am sure new adopters would appreciate it a lot.

smaillet-ms, 2015-08-04:
The newer APIs have some comments, now that the actual Windows 10 APIs and SDK are available we'll be updating comments to match (See #162 ) Documentation is a large task and we are still catching up on technical debt from previous releases and time. It's also an endless job as no documentation set is enough or right for everyone, there's always room to improve. As always, since we are an Open Source project, we are more than willing to take contributions that can help in this area.

dimso, 2015-08-04:
That's why I think that making visible already available comments would be useful. Something is better than nothing, so why not generate xml files and make those comments visible?

smaillet-ms, 2015-08-04:
Historical docs process really. The build supports generating the XML files but the SDK installer doesn't have support to include the XML files. That's not rocket science to do, but it is a tedious pain that hasn't made it up the priority list - it's a perfect place for contributions though - hint hint :smile: