NETMF/netmf-interpreter #270: Build MF v4.4 on clean x64 machine?

doingnz, 2015-08-05:
I am failing to build vanilla MF v4.4. Appreciate any suggestions what I might have done incorrectly.

I believe I have followed the initial 6 steps in the Getting Started correctly and have a local copy of netmf-interpreter, CMSIS & build-tools. I have checked the VS2015 setup and have nearly everything installed.

I note that unzipping the "build-tools.zip" into the netmf-interpreter folder adds a x86 folder to tools. 

My Virtual machine is running Windows 10 in x64 mode. Do I need x64 versions of the items in x86 folder? 

See errors below: Note that CODESIGN.Submitter.dll is in the folder "C:\Projects\BP+\netmf\netmf-interpreter\build-tools.zip\tools\x86\CODESIGN"

What sort of command window should I open to run the build_sdk.cmd? I have tried a few with the same results each time. 

I must have something simple wrong in my setup. Appreciate a pointer as to what might be wrong.

When Build_sdk.cmd executes, the following errors are displayed.

C:\Projects\BP+\netmf\netmf-interpreter>build_sdk
Compiler: VS 14
Building PreSDK ...
Microsoft (R) Build Engine version 14.0.23107.0
Copyright (C) Microsoft Corporation. All rights reserved.

  WixLibSpotBuild -> C:\Projects\BP+\netmf\netmf-interpreter\BuildOutput\public\Release\Server\dll\Microsoft.SPOT.WiX.d
  ll
C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets(1819,5): warning MSB3245: Could not res
olve this reference. Could not locate the assembly "CODESIGN.Submitter". Check to make sure the assembly exists on disk
. If this reference is required by your code, you may get compilation errors. [C:\Projects\BP+\netmf\netmf-interpreter\
Framework\Tools\BuildTasksInternal\BuildSigner\BuildSignerSpotBuild.csproj]
BuildSigner.cs(7,7): error CS0246: The type or namespace name 'CODESIGN' could not be found (are you missing a using di
rective or an assembly reference?) [C:\Projects\BP+\netmf\netmf-interpreter\Framework\Tools\BuildTasksInternal\BuildSig
ner\BuildSignerSpotBuild.csproj]
C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets(1819,5): warning MSB3245: Could not res
olve this reference. Could not locate the assembly "wix". Check to make sure the assembly exists on disk. If this refer
ence is required by your code, you may get compilation errors. [C:\Projects\BP+\netmf\netmf-interpreter\Framework\Tools
\BuildTasksInternal\InternalBuildTasks.csproj]
C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets(1819,5): warning MSB3245: Could not res
olve this reference. Could not locate the assembly "Microsoft.SPOT.AutomatedBuild.BuildSigner". Check to make sure the
assembly exists on disk. If this reference is required by your code, you may get compilation errors. [C:\Projects\BP+\n
etmf\netmf-interpreter\Framework\Tools\BuildTasksInternal\InternalBuildTasks.csproj]
SignMSI.cs(6,22): error CS0234: The type or namespace name 'AutomatedBuild' does not exist in the namespace 'Microsoft.
SPOT' (are you missing an assembly reference?) [C:\Projects\BP+\netmf\netmf-interpreter\Framework\Tools\BuildTasksInter
nal\InternalBuildTasks.csproj]
SignMSIAssemblies.cs(6,22): error CS0234: The type or namespace name 'AutomatedBuild' does not exist in the namespace '
Microsoft.SPOT' (are you missing an assembly reference?) [C:\Projects\BP+\netmf\netmf-interpreter\Framework\Tools\Build
TasksInternal\InternalBuildTasks.csproj]
CreatePseudoInstallScript.cs(11,17): error CS0234: The type or namespace name 'Tools' does not exist in the namespace '
Microsoft' (are you missing an assembly reference?) [C:\Projects\BP+\netmf\netmf-interpreter\Framework\Tools\BuildTasks
Internal\InternalBuildTasks.csproj]
CreatePseudoInstallScript.cs(12,17): error CS0234: The type or namespace name 'Tools' does not exist in the namespace '
Microsoft' (are you missing an assembly reference?) [C:\Projects\BP+\netmf\netmf-interpreter\Framework\Tools\BuildTasks
Internal\InternalBuildTasks.csproj]
SignMSI.cs(16,17): error CS0246: The type or namespace name 'SignInfo' could not be found (are you missing a using dire
ctive or an assembly reference?) [C:\Projects\BP+\netmf\netmf-interpreter\Framework\Tools\BuildTasksInternal\InternalBu
ildTasks.csproj]
SignMSIAssemblies.cs(17,17): error CS0246: The type or namespace name 'SignInfo' could not be found (are you missing a
using directive or an assembly reference?) [C:\Projects\BP+\netmf\netmf-interpreter\Framework\Tools\BuildTasksInternal\
InternalBuildTasks.csproj]
CreatePseudoInstallScript.cs(20,17): error CS0246: The type or namespace name 'Database' could not be found (are you mi
ssing a using directive or an assembly reference?) [C:\Projects\BP+\netmf\netmf-interpreter\Framework\Tools\BuildTasksI
nternal\InternalBuildTasks.csproj]
C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets(1819,5): warning MSB3245: Could not res
olve this reference. Could not locate the assembly "Microsoft.SPOT.Tasks.Internal". Check to make sure the assembly exi
sts on disk. If this reference is required by your code, you may get compilation errors. [C:\Projects\BP+\netmf\netmf-i
nterpreter\Framework\Tools\BuildTasks\BuildTasks.csproj]
Attempting to cancel the build...
C:\Projects\BP+\netmf\netmf-interpreter\tools\Targets\Microsoft.SPOT.Build.Common.Targets(96,5): error MSB3202: The pro
ject file "Framework\IDE\build.dirproj" was not found. [C:\Projects\BP+\netmf\netmf-interpreter\sdk.dirproj]
C:\Projects\BP+\netmf\netmf-interpreter\tools\Targets\Microsoft.SPOT.Build.Common.Targets(96,5): error MSB3202: The pro
ject file "Product\AllSDK\build.dirproj" was not found. [C:\Projects\BP+\netmf\netmf-interpreter\sdk.dirproj]
C:\Projects\BP+\netmf\netmf-interpreter\tools\Targets\Microsoft.SPOT.Build.Common.Targets(96,5): error MSB3202: The pro
ject file "Product\Samples\build.dirproj" was not found. [C:\Projects\BP+\netmf\netmf-interpreter\sdk.dirproj]
C:\Projects\BP+\netmf\netmf-interpreter\tools\Targets\Microsoft.SPOT.Build.Common.Targets(96,5): error MSB3202: The pro
ject file "CLR\Tools\TFConvert\Fonts\TrueType\TrueTypeFonts.wixproj" was not found. [C:\Projects\BP+\netmf\netmf-interp
reter\sdk.dirproj]

doingnz, 2015-08-05:
The last entry in visixpkg.log says "warning NETMF: Visual Studio 2013 SDK not found VSIX package will not be generated". Maybe I misunderstood that earlier issues resolved this and that VS2013 was no longer required to be installed?

Sorry if these are dumb newbie questions.

visixpkg.log is:
```
Build started 5/08/2015 2:28:09 PM.
Overriding target "GetFrameworkPaths" in project "C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets" with target "GetFrameworkPaths" from project "C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.NetFramework.CurrentVersion.targets".
Overriding target "SatelliteDllsProjectOutputGroup" in project "C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets" with target "SatelliteDllsProjectOutputGroup" from project "C:\Windows\Microsoft.NET\Framework\v4.0.30319\Microsoft.WinFx.targets".
Overriding target "IncrementalClean" in project "C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets" with target "IncrementalClean" from project "C:\Projects\BP+\netmf\netmf-interpreter\tools\Targets\Microsoft.SPOT.Build.Common.Targets".
Overriding target "_CleanGetCurrentAndPriorFileWrites" in project "C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets" with target "_CleanGetCurrentAndPriorFileWrites" from project "C:\Projects\BP+\netmf\netmf-interpreter\tools\Targets\Microsoft.SPOT.Build.Common.Targets".
Overriding target "CoreClean" in project "C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets" with target "CoreClean" from project "C:\Projects\BP+\netmf\netmf-interpreter\tools\Targets\Microsoft.SPOT.Build.Common.Targets".
Overriding target "Compile" in project "C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets" with target "Compile" from project "C:\Projects\BP+\netmf\netmf-interpreter\tools\Targets\Microsoft.SPOT.Targets".
Overriding target "CopyFilesToOutputDirectory" in project "C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets" with target "CopyFilesToOutputDirectory" from project "C:\Projects\BP+\netmf\netmf-interpreter\tools\Targets\Microsoft.SPOT.Targets".
Overriding target "GetFrameworkPaths" in project "C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.NetFramework.CurrentVersion.targets" with target "GetFrameworkPaths" from project "C:\Projects\BP+\netmf\netmf-interpreter\tools\Targets\Microsoft.SPOT.Targets".
Overriding target "GetReferenceAssemblyPaths" in project "C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets" with target "GetReferenceAssemblyPaths" from project "C:\Projects\BP+\netmf\netmf-interpreter\tools\Targets\Microsoft.SPOT.Targets".
The target "AfterGenerateAppxManifest" listed in an AfterTargets attribute at "C:\Program Files (x86)\MSBuild\Microsoft\.NetNative\Microsoft.Net.CoreRuntime.targets (68,11)" does not exist in the project, and will be ignored.
The target "_GeneratePrisForPortableLibraries" listed in a BeforeTargets attribute at "C:\Program Files (x86)\MSBuild\Microsoft\.NetNative\Microsoft.Net.CoreRuntime.targets (177,11)" does not exist in the project, and will be ignored.
The target "AfterGenerateAppxManifest" listed in an AfterTargets attribute at "C:\Program Files (x86)\MSBuild\Microsoft\.NetNative\Microsoft.NetNative.targets (126,11)" does not exist in the project, and will be ignored.
The target "AfterGenerateAppxManifest" listed in an AfterTargets attribute at "C:\Program Files (x86)\MSBuild\Microsoft\.NetNative\Microsoft.NetNative.targets (174,11)" does not exist in the project, and will be ignored.
The target "BeforeGenerateProjectPriFile" listed in a BeforeTargets attribute at "C:\Program Files (x86)\MSBuild\Microsoft\NuGet\Microsoft.NuGet.targets (149,61)" does not exist in the project, and will be ignored.
Project "C:\Projects\BP+\netmf\netmf-interpreter\setup\ProductSDK\VsixPackages.dirproj" on node 1 (Build target(s)).
Building with tools version "14.0".
Project file contains ToolsVersion="4.0". This toolset may be unknown or missing, in which case you may be able to resolve this by installing the appropriate version of MSBuild, or the build may have been forced to a particular ToolsVersion for policy reasons. Treating the project as if it had ToolsVersion="14.0". For more information, please see http://go.microsoft.com/fwlink/?LinkId=293424.
Target "CheckForPrerequisites" in project "C:\Projects\BP+\netmf\netmf-interpreter\setup\ProductSDK\VsixPackages.dirproj" (entry point):
Using "Warning" task from assembly "Microsoft.Build.Tasks.Core, Version=14.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a".
Task "Warning"
C:\Projects\BP+\netmf\netmf-interpreter\setup\ProductSDK\VsixPackages.dirproj(82,9): warning NETMF: Visual Studio 2013 SDK not found VSIX package will not be generated
Done executing task "Warning".
Task "Warning" skipped, due to false condition; (!Exists('$(VSSDK14INSTALLDIR)')) was evaluated as (!Exists('C:\Program Files (x86)\Microsoft Visual Studio 14.0\VSSDK\')).
Done building target "CheckForPrerequisites" in project "VsixPackages.dirproj".
Target "_CheckForInvalidConfigurationAndPlatform" in file "C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets" from project "C:\Projects\BP+\netmf\netmf-interpreter\setup\ProductSDK\VsixPackages.dirproj" (entry point):
[... cut ...]
Done Building Project "C:\Projects\BP+\netmf\netmf-interpreter\setup\ProductSDK\VsixPackages.dirproj" (Build target(s)) -- FAILED.

Build FAILED.

"C:\Projects\BP+\netmf\netmf-interpreter\setup\ProductSDK\VsixPackages.dirproj" (Build target) (1) ->
(CheckForPrerequisites target) -> 
  C:\Projects\BP+\netmf\netmf-interpreter\setup\ProductSDK\VsixPackages.dirproj(82,9): warning NETMF: Visual Studio 2013 SDK not found VSIX package will not be generated

    1 Warning(s)
    0 Error(s)

Time Elapsed 00:00:00.12
```

SytechDesigns, 2015-08-05:
I have been working thru similar problems. Your initial problem is the location of the build tools.zip.
Assuming you have VS2015 installed with C++ compiler ( in release version this is not installed in the 'typical' install, need to use custom install and select the c++ tools), and also the vs2015 SDK is installed.
You need to unzip buildtools into a directory folder above netmf-interpreter. not in the net_mf-interpreter\tools directory.
i.e if you clone locally into :
c:\git_repo\netmf-interpreter - then build tools needs to be unzipped into c:\git_repo, so you end up with:
C:\git_repo
                  bin
                  netmf-interpreter
                  tools

The references in the projects for things like CODESIGN.Submitter are SPOROOT\Tools\X86..
where SPOROOT is the parent folder of SPOCLIENT ( netmf-interpreter in this case).

Your second point about VS2013 vsix is, if you have VS2013/SDK installed, then a vsix will be generated for vs2013 as well, if it is not installed then this step will be skipped, but you must have vs2015 installed or you will fail the first check in build-sdk

SytechDesigns, 2015-08-05:
Directory structure did not come out as clear as I intended in above post - so to clarify
C:\git_repo\bin
C:\git_repo\netmf-interpreter
C:\git_repo\tools

doingnz, 2015-08-05:
Thank you! The Build is now running fine! Awesome

doingnz, 2015-08-07:
When doing a clean build_sdk, the following warning is raised. Is this a concern?
```
  Linking C:\Projects\BPplus\netmf\netmf-interpreter\BuildOutput\Windows\le\ANY_MEDIA\release\lib\Xml.lib
ParseOptions_Win32.cpp(296): warning C4474: 'wprintf' : too many arguments passed for format string [C:\Projects\BPplus
\netmf\netmf-interpreter\CLR\Core\dotnetmf.proj]
  Linking C:\Projects\BPplus\netmf\netmf-interpreter\BuildOutput\Windows\le\ANY_MEDIA\release\lib\Core.lib
```

doingnz, 2015-08-09:
I am closing this issue as the final question is already covered in #257

closed 2015-08-09 by doingnz