NETMF/netmf-interpreter #272: Linking external code artifacts

DerekGn, 2015-08-05:
I am working on a port at the moment. I have code artifacts created by the hardware vendor. I am wondering what is the best way to utilize these code artifacts.

For example I have a bootstrap function that I can reuse for my target device port.

```
void BootstrapCode()
{
	LowLevelInit();
}
```
If I simply include the source file in the project file for the bootstrap code

```
<Compile Include="board_lowlevel.c" />
<Compile Include="SAMA5D3_bootstrap.cpp" />
```
The module compiles correctly once the settings file has the relevant header includes etc.

The .obj file is created in the correct build output folder, but the linker cannot find the definition for the function LowLevelInit

  SAMA5D3_bootstrap.lib(SAMA5D3_bootstrap.obj): In function `BootstrapCode':
  SAMA5D3_bootstrap.cpp:(.text.BootstrapCode+0x10): undefined reference to `LowLevelInit()'

If i check the manifest for the SAMA5D3_bootstrap.lib, it seems to indicate that the obj code is included.

````
<ItemGroup>
    <ObjectFile Include="D:\GitHub\NetMf\netmf-interpreter\BuildOutput\ARMv7-AR\GCC4.9\le\RAM\release\SAMA5D3XPLN\obj\DeviceCode\Targets\Native\SAMA5D3\DeviceCode\SAMA5D3_Bootstrap\board_lowlevel.obj" />
    <ObjectFile Include="D:\GitHub\NetMf\netmf-interpreter\BuildOutput\ARMv7-AR\GCC4.9\le\RAM\release\SAMA5D3XPLN\obj\DeviceCode\Targets\Native\SAMA5D3\DeviceCode\SAMA5D3_Bootstrap\SAMA5D3_bootstrap.obj" />
</ItemGroup>
```

if I dump the SAMA5D3_bootstrap.lib file i can see the functions

board_lowlevel.obj:
         U BOARD_RemapRam
00000000 T defaultFiqHandler
00000000 T defaultIrqHandler
00000000 T defaultSpuriousHandler
00000000 W LowLevelInit
         U PMC_SelectExt12M_Osc
         U PMC_SetMckDivider
         U PMC_SetMckPllaDiv
         U PMC_SetMckPrescaler
         U PMC_SetPllA
         U PMC_SwitchMck2Main
         U PMC_SwitchMck2Pll

SAMA5D3_bootstrap.obj:
         U _Z12LowLevelInitv
00000000 T BootstrapCode

I am looking for some hints as to what I am missing with this specific issue, I am a bit rusty with some of this stuff, secondly what is the correct way to include external code in the .net mf interpreter?

DerekGn, 2015-08-05:
I think I have fumbled my way to a solution as usual 30 seconds after asking a question for help....

It looks like its a compilation flag/options issue. If I rename file to .cpp the code is compiled and linked correctly.

I will need to check what flag/option will allow linking without having to rename files. I would prefer too leave the vendor code as intact as possible.

closed 2015-11-03 by DerekGn

smaillet-ms, 2015-11-03:
The likely cause is missing extern "C" checks in header files when built in C++, since C++ allows overloaded functions it creates mangled function names for the linker based on the function name, scope, return type and arg types. There's no universal standard for mangling but C++ supports the extern "C" syntax to effectively disable the mangling (and overload support for that function)

DerekGn, 2015-11-04:
Yes, I resolved the problem by creating a container header file that included the headers for the c code libs.

```
extern "C" {
	#include "existingc.h"
}
```