NETMF/netmf-interpreter #275: STM32F4: Incorrect hardware configuration

cw2, 2015-08-05:
The current implementation uses `STM32F4XX ` preprocessor symbol to setup various hardware configuration options. However, it was meant for 40x variants and some new chips (e.g. 411) require different setup - for example, they do not have CCM, have only 3 USARTs etc. Also, this symbol is used to distinguish between F2 and F4 series (#include).

IMHO symbols from `stm32f4xx.h`, such as `STM32F40_41xxx` or `STM32F429_439xx` should be used instead.

Or, what is the time-frame for mbed libraries?

smaillet-ms, 2015-08-05:
yes there are still some legacy constructs here. Ideally we'd want to use the current CMSIS header files instead of older cloned copies. In addition, we should be leveraging the existing silicon vendor libraries as-is as much as possible so we don't get into issues created from porting code and then not getting bug fixes etc...
The [mBed site](http://mbed.org/technology/os/) lists This month as the first Beta release of mBed 3.0. When that is public we can discuss how we want to work with that code with NETMF.

josesimoes, 2015-08-06:
I agree with cw2 request. Working with several variants (and more coming!) requires a more granular definition of the variant.

cw2, 2015-08-17:
Now when people start asking about mbed 3.0 beta release
> I expect the release to be later in the month. - Stephen Paulger

https://developer.mbed.org/questions/55571/mbed-30-beta-release/