NETMF/netmf-interpreter #278: SolutionWizard no longer in tools

SytechDesigns, 2015-08-09:
Solution Wizard is missing in the MF4.4 tools. Will there be a new tool for cloning solutions or will Solution Wizard be added back to tools. At the moment I am using Solution Wizard from MF4.3 to clone solutions.

JohnySeven, 2015-08-09:
Hi, check this pull request: https://github.com/NETMF/netmf-interpreter/pull/122

SytechDesigns, 2015-08-09:
Thanks for the reference - I see, but as in #249, we do need some sort of tool for cloning solutions. For now I will use the old 4.3 version - or maybe even go back to some old power shell scripts we used to use to clone solutions. Solution Wizard could be a 'odd',  but it does give a reasonable base to edit for the new clone.

smaillet-ms, 2015-08-10:
Unfortunately, with the inclusion of the RTOS to fix multiple LWIP stack issues solution wizard is just broken beyond repair. To properly handle all the "special cases" required would mean a massive re-design. Though we'd welcome any suggestions or proposals on designs for a simpler solution to get a basic clone going.

cw2, 2015-08-11:
I've cloned a few solutions recently and as long as there is a good source, it is not that hard - a few tips to make it a little bit easier:
* Order driver libraries (\<ItemGroup>\<xxxLibs.../>\<Required Projects... />\</ItemGroup>) alphabetically in .proj file. Its ~20 lines of C# or similar and makes it really easy to copy and most importantly to compare those files.
* Similarly (a few lines) to generate/update new GUIDs (\<ProjectGuid>),
* Replace \<ProjectPath> with $(MSBuildThisFileFullPath) to remove hardcoded path,
* Delete unused elements that contain solution name - like \<CustomSpecific>

IMHO those two last items applied en-masse would reduce the cloning work to a minimum, most of the build variables (like Directory, various Solution-base ones etc.) could be either defined relative to the project file or based on (one) value in the main .proj or .settings file. It should be enough to define only the solution name.

lt72, 2015-08-11:
Cleanup is definitely something we should be doing in the background and I agree that it can go a long way. I also think that having a white paper could be helpful... or a wiki.

cw2, 2015-08-11:
Well, I have a few things already on my list (that started long time ago), which I am more than willing to contribute - I guess I could start in a dedicated 'cleanup' branch with small/simple deletes in mostly random areas and discuss bigger tasks in some 'Cleanup' topic/issue... (?)