NETMF/netmf-interpreter #281: generics support

wizzard0, 2015-08-10:
Is there a chance generics will eventually be supported on netmf?

Any option, eg CIL preprocessing, RTTI or new LLVM pipeline will do. Just ease the porting - this is the biggest hurdle encountered so far.

smaillet-ms, 2015-08-10:
For the interpreter, not likely (It actually decreases performance and increases code size). However, generics is a key part of what's driving us to Ahead of Time compilation with the [Llilum project](https://github.com/NETMF/llilum), while that is still in the very early stages you can watch our progress and provide feedback as we go.

wizzard0, 2015-09-18:
buuut llilum IIRC will not support dynamic assembly loading etc, so it is not a clear "win-win" option :)

also - again, i'm fine with performance hit, just enable faster porting desktop .NET code, with option to optimize it later...

mpielikis, 2015-10-18:
I made a tool that replaces generics with casts to object. It's just a prove of concept, but it could help sometimes. https://github.com/mpielikis/retrosharp

wizzard0, 2015-10-18:
Oh, great! Will try to use it at some point