NETMF/netmf-interpreter #286: C# fixed buffer fields

Korporal, 2015-08-24:
I'd assumed that because .NET MF targets hardware and devices, that it would naturally support C# fixed buffers, this language feature makes it easy to create contiguous structures with embedded inline arrays and so on - ideal for hardware applications.

https://msdn.microsoft.com/en-us/library/zycewsya.aspx

Sadly it seems to be unsupported due to the absence of the FixedBufferAttributre class which the C# compiler uses to decorate fixed buffer fields.

Is this something that's been simply overlooked and could be supported or something that was intentionally left out?

Thanks.

smaillet-ms, 2015-08-24:
It was deliberately left out. It is an unsafe construct and a deliberate design choice of NETMF was to eliminate all unsafe support. Not only for safety/security but because mapping of data structures with the fixed keyword assumes the internal structures are actually compatible with a "C" style layout, which because NETMF is an interpreter isn't how things end up in memory. This would cause extensive copy marshalling, which essentially defeats the purpose of creating such mapping to begin with. With the [Llilum Project](https://github.com/NETMF/llilum) we'll be supporting this sort of functionality and then some. (Ideally being able to do really deep systems level programming direct on the metal with little or no use of the "unsafe" keyword required )

Korporal, 2015-08-29:
OK so this is by design, fair enough. Thanks.

closed 2015-08-29 by Korporal