NETMF/netmf-interpreter #289: Mqtt stop publishing after a few posts

justbwalker, 2015-08-28:
Hi.

I´m testing the new tcp/ip stack on NETMF 4.4 with Paolo Patierno's M2Mqtt library, it works fine, but after a few publishes it stop posting the messages. The number of publishs before it stops is always the same if we send the same message lenght. If I send a shorter messages it sends more times before stop and if I send longer messages it stops less times.

The code still running without any exceptions and prints a test string in the debug console like it stills sending the message but no data came out from de Netduino.

What can be happening ? Somebody can help me ?

Thanks !

ppatierno, 2015-08-29:
Hi,

until today I had no chance to try my M2Mqtt library on the 4.4 version. However, I know about the embedded MVP Lorenzo Maiorfi used it without any problems or relevant feedbacks. What's the board you are working on ?

Thanks,
Paolo.

NicolasBesson, 2015-08-29:
OK I hope that you will be able to have your lib compliant with 4.4.
My new short project is based on a microcontroller, I don't have any OS, it
is an Atmel attiny with 8k of flash and 512B of Ram :-)

justbwalker, 2015-08-29:
Sorry, I was wrong. Im using the bulid 6 of netmf 4.3.2.2 with netduino.ip.
It wasn't the 4.4.
I'm using Netduino Plus 2.

maiorfi, 2015-08-30:
Hi.

I've run a simple mqtt echo test with a remote (azure hosted) broker at about 1 msg/s rate on reference Keil board running .netmf 4.4 (as per state it had a couple of months ago on github) and it kept running fine for several days in a row without hiccups.

I have also several successfully running applications in production (i.e. 24/7) running on .netmf 4.3.1 using M2Mqtt library.

Could you share some code I could run on Keil board?

ppatierno, 2015-08-30:
So it seems that the problem is related to the Netduino.IP stack that is a completely managed stack developed by Netduino team in C#.

Paolo.