NETMF/netmf-interpreter #290: Fixed width fonts.

Korporal, 2015-08-29:
I've been struggling to get a fixed width font working with the Bitmap class. I created several (supposedly) fixed width .tinyfnt files from Lucida Console, Courier New and Consolas. Howewever the computed text extend isn't what we'd expect if the font were fixed width and test text shows that its not fixed with when rendered.

I played a little with the various DT_ flags but to no avail. Is there a bug perhaps in the TFConvert utility that somehow mishandles fixed width fonts?

miloush, 2015-08-30:
Please note the latest TFTool GUI no longer uses TFConvert.

You say that _the computed text extents for various test strings yields different values for (computed_extent / length_of_string)._ Can you give a particular example of font, size (or export definition), failing test string and results?

Korporal, 2015-08-30:
Hi, here's some simple test code I added to observe under debug:
```
            
            font.ComputeExtent("Hello", out w, out h, 1);
            font.ComputeExtent("Hello Im some iiii code", out w, out h, 1);
            font.ComputeExtent("Hello Im some iiii code with a time of 08:23:17", out w, out h, 1);
            font.ComputeTextInRect("Hello", out w, out h, 0, 0, displayWidth, displayHeight, Bitmap.DT_AlignmentCenter);
            font.ComputeTextInRect("Hello Im some iiii code", out w, out h, 0, 0, displayWidth, displayHeight, Bitmap.DT_AlignmentMask);
            font.ComputeTextInRect("Hello Im some iiii code with a time of 08:23:17", out w, out h, 0, 0, displayWidth, displayHeight, Bitmap.DT_AlignmentMask);   
```
You can observe the 'w' parameter after each call and with the length of each string you can see the extent varies so (extent of text / length of string) is not constant.

I created three fonts using the GUI tool, all were sized as 12. They were Courier New, Consolas and Lucida Console.

Another simple test is to just display some mixed strings each the same length and see the extent on the display varies - some are visibly shorter than others.

I'm attaching a screen shot too, all lines contain the same number of characters - 30.

[image]

I'll now run TFConvert directly see what happens...

Korporal, 2015-08-30:
Just a note: I think it would be helpful for Microsoft to actually include some sample .fntdef files to help in understanding the process. Also why does NETMF include only TWO sample fonts? Microsoft have hundreds of fonts available in Windows which we use to create desktop app, why couyldn't they inc;ude a bigger set AND include at least one fixed width?

miloush, 2015-08-30:
Probably could, I guess no one asked yet.

The TFConvert resp. .fntdef format is documented at https://msdn.microsoft.com/en-us/library/cc533020.aspx, includes samples now and then. Exporting definitions from the GUI tool should also give you some idea. The problem is definitely not in the definition, you can use it as is.

I did repro the problem with + and -, seems the reported glyph bearings and ink box widths do not add to the same advance width. I will try to look into what I can do about it.

Korporal, 2015-08-30:
This is all very kludgy, I've created a .fntdef file, have a .ttf file yet when I run tfconvert it runs, reports nothing at all and there's no output file.

Korporal, 2015-08-30:
OK looking good. I didnt realize your tool could export the .fntdef (which it suffizes as .fontdef).  Anyway  I did that and used TFConvert to to create the .tinyfnt and reran my test, it is now fixed width.

Korporal, 2015-08-30:
Incidentally, is there any sort of test one can do on a Font instance, to verify it is indeed a fixed width font? or must I compare the widths of some set of characters to ensure there the same and then assume its a true fixed width font?

miloush, 2015-08-30:
Re: fntdef vs fontdef - good catch, will fix that too.

If you mean on the MF device, than no you need to compare widths. If you are asking about desktop, this is a bit tricky. Technically, fonts can claim whether they are fixed width or not, but this is just a flag set by the font designer and does not necessarily correspond to the reality. You are also asking for a font to be globally fixed width, though it might be only for some scripts. I believe the best and easiest way is to check the width of i vs w glyphs, see http://www.siao2.com/2012/08/14/10338893.aspx.

Korporal, 2015-08-30:
OK, I'll rely on a simple heuristic to check for fixed width. Also the actual char width for the fixed width font is obtained from Font.CharWidth() - for any char, and seems to match exactly with the extent computation methods.

Thanks for helping out - my mini-project is now back in the saddle!